Flipper-XFW/Xtreme-Firmware #25: build/f7-firmware-C/toolbox/version.inc.h: Error 1

sudo001, 2022-12-24:
### Describe the bug.

Whenever i try to build from source using ./fbt updater_package, it throws:

`********** FBT ERRORS **********
build/f7-firmware-C/toolbox/version.inc.h: Error 1
`

### Reproduction

‎

### Target

Compile Error

ClaraCrazy, 2022-12-24:
Adding logs would really help.

Luckily, I seen it on OFW. Anyways. linux building has been broken on other CFWs and OFW for days, I don't know why since I haven't fired up a linux box in a while and don't really have the time to either as u know.. christmas is a thing. 

For now all I can recommend is using Windows to build. Appreciate the info tho, albeit very sparse for some reason. I'll see if I can get one of my ~~slaves~~ devs to work on it, lol

Merry christmas

ClaraCrazy, 2022-12-24:
I cant leave issues un-fixed... Add me on discord "Magical Unicorn#8161" don't have a server anymore, as discord terminated mine, and probably will do the same for other CFWs

ClaraCrazy, 2022-12-24:
the command that errors out is a git command, which has nothing to do with any firmware details. Seems like a miss-configured git client

ClaraCrazy, 2022-12-24:
-- closed as of now --

Issue is originating from a git command, nothing (C)FW related.

closed 2022-12-24 by ClaraCrazy