Flipper-XFW/Xtreme-Firmware #33: Bluetooth update / cannot find directory / preload failed

DaveC397, 2022-12-25:
### Describe the bug.

Can not install from qflipper phone app "unable to install something went wrong"

Can not install tgz (corrupted) with qflipper desktop windows 10

Once installed using dfu file in windows qflipper clock does not work. 

Qflipper for windows 64bit shows dev: unknown

Clock will not load. "Error preload failed API version mismatch

### Reproduction

Windows 10 machine newest qflipper 64bit

### Logs

```Text
1205 [APP] qFlipper version 1.2.2 commit ba67025d 2022-12-08T07:31:11
1205 [APP] OS info: Windows 10 Version 2009 10 10.0.19045 Qt 5.15.2
1568 [UPD] Fetched update information from https://update.flipperzero.one/qFlipper/directory.json
1568 [UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
23204 [REG] Detected new device: VID_0x483:PID_0x5740
23417 [DBG] Trying serial port FLIP_L0MB4T at \\.\COM3
23417 [DBG] Using  serial port FLIP_L0MB4T at \\.\COM3
23417 [RPC] Starting RPC session...
23624 [RPC] RPC session started successfully.
23625 [RPC] (1) System Protobuf Version START
23682 [DBG] Detected protobuf version: 0.14
23682 [RPC] (1) System Protobuf Version SUCCESS
23691 [RPC] (2) Property Get START
23808 [RPC] (2) Property Get SUCCESS
23816 [RPC] (3) Storage Info @/ext START
23874 [RPC] (3) Storage Info @/ext SUCCESS
23882 [RPC] (4) Storage Stat @/ext/Manifest START
23941 [RPC] (4) Storage Stat @/ext/Manifest SUCCESS
23941 [RPC] (5) System Get DateTime START
24001 [DBG] Flipper time skew is -2045 milliseconds
24002 [RPC] (5) System Get DateTime SUCCESS
24008 [RPC] (6) System Set DateTime START
24067 [RPC] (6) System Set DateTime SUCCESS
24083 [RPC] Stopping RPC session...
24097 [RPC] Attempting to unload protobuf plugin...
24097 [RPC] Unloaded protobuf plugin.
24097 [RPC] RPC session stopped successfully.
24098 [REG] Registering the device
24098 [DEV] Version: 0.73.2 commit: 20a0a43d2 radio: 1.13.3
24098 [BKD] Current device changed to "L0mb4t"
24198 [RPC] Starting RPC session...
24274 [RPC] RPC session started successfully.
24283 [RPC] (1) Gui Start ScreenStream START
24341 [RPC] (1) Gui Start ScreenStream SUCCESS
355652 [DEV] Full Update @L0mb4t START
355661 [RPC] (2) Gui Start VirtualDisplay START
355667 [UTL] Region Provisioning @L0mb4t START
355667 [DBG] Development hardware region detected, skipping region provisioning...
355668 [UTL] Region Provisioning @L0mb4t SUCCESS
355685 [UTL] Refresh Storage Info @L0mb4t START
355734 [RPC] (2) Gui Start VirtualDisplay SUCCESS
355743 [RPC] (3) Storage Info @/ext START
355801 [RPC] (3) Storage Info @/ext SUCCESS
355812 [RPC] (4) Storage Stat @/ext/Manifest START
355867 [RPC] (4) Storage Stat @/ext/Manifest SUCCESS
355928 [UTL] Refresh Storage Info @L0mb4t SUCCESS
355934 [ZIP] Uncompressing file with size of 2257072 bytes...
355993 [ZIP] Uncompression finished : No Error
356010 [DEV] Full Update @L0mb4t ERROR: Cannot find update directory
356010 [BKD] Current operation finished with error: "Cannot find update directory"
366484 [BKD] Finalized current operation
[... 546 lines cut ...]
1019462 [RPC] RPC session stopped successfully.
1019462 [REG] Registering the device
1019462 [DEV] Version: 0.74.2 commit: Unknown radio: 1.13.3
1019462 [BKD] Current device changed to "L0mb4t"
1019568 [RPC] Starting RPC session...
1019656 [RPC] RPC session started successfully.
1019657 [RPC] (1) Gui Start ScreenStream START
1019720 [RPC] (1) Gui Start ScreenStream SUCCESS
1140536 [DEV] Full Update @L0mb4t START
1140543 [RPC] (2) Gui Start VirtualDisplay START
1140552 [UTL] Region Provisioning @L0mb4t START
1140560 [DBG] Development hardware region detected, skipping region provisioning...
1140568 [UTL] Region Provisioning @L0mb4t SUCCESS
1140585 [UTL] Refresh Storage Info @L0mb4t START
1140601 [RPC] (2) Gui Start VirtualDisplay SUCCESS
1140610 [RPC] (3) Storage Info @/ext START
1140668 [RPC] (3) Storage Info @/ext SUCCESS
1140668 [RPC] (4) Storage Stat @/ext/Manifest START
1140727 [RPC] (4) Storage Stat @/ext/Manifest SUCCESS
1140786 [UTL] Refresh Storage Info @L0mb4t SUCCESS
1140814 [ZIP] Uncompressing file with size of 2257072 bytes...
1140868 [ZIP] Uncompression finished : No Error
1140887 [DEV] Full Update @L0mb4t ERROR: Cannot find update directory
1140887 [BKD] Current operation finished with error: "Cannot find update directory"
```

L0rdK0nda, 2022-12-25:
The .tgz should be fixed next update

ClaraCrazy, 2022-12-25:
Hey

As for the error, "Cannot find update directory" makes it sound like your flipper doesnt have that, which makes it clear why no update is working. Either way, our .tgz file is broken bc I had to hacksmith that together at like 5am which was a bad idea

For the apps with API mismatch, that's something we really gotta bring up in OFW. Updating firmware does not update all apps by default. Our firmware uses a higher API version and old apps don't run. Easy fix, remove your apps folder before the update (using the .zip with manual update for now, I apologize). That fixes apps. Gotta make an issue about that on OFW and add it to our readme.

Thanks for bringing this to our attention and ofc for using our firmware. Feel free to join the discord too!

closed 2022-12-25 by ClaraCrazy