Flipper-XFW/Xtreme-Firmware #37: Give credits.

ankris812, 2022-12-27:
### Description of the feature you're suggesting.

If you are going to take Code from other people, why dont you give credits? The gpio and ir replication is from my firmware.

ClaraCrazy, 2022-12-27:
I have no idea what makes you think I took code from you, since I have literally never heard of you... not to mention that after a quick glance your two flipper forks don't even have unique commits, but uh... sure buddy!

ankris812, 2022-12-27:
If It wasnt you, then it was Roguemaster. Sorry if its the case.

ClaraCrazy, 2022-12-27:
No, actually lets keep this open. I'm not the only one working on this and maybe should take that into account too. My points stand but feel free to link your code so I can handle this if that's true

ClaraCrazy, 2022-12-27:
Dont want any negative views here. If its yours, drop a link and I'll have it handled asap

ankris812, 2022-12-27:
Sure. Just let me find when i made the commit. And just as a note. Signal replication doesnt work correctly on the subghz part. Signal output is a bit distorted.

ClaraCrazy, 2022-12-27:
Thanks, and noted. Might have one of my devs look over it perhaps

ankris812, 2022-12-27:
Also you have this https://github.com/ankris812/Yggdrassil/commit/f8aa139b117c3f9db89a60892eafb1ef02a0f26e If you want to add It to the resources.

ClaraCrazy, 2022-12-27:
Good to know where those came from. Only downloaded them from a discord lol

ankris812, 2022-12-27:
Lately I don't have time for my firmware, but if it's not a bother, I think I'll collaborate from time to time with this firmware, it fits exactly what I want to do with mine, something more advanced and functional, but that it remains stable.

ClaraCrazy, 2022-12-27:
sure sure. would be nice if you'd join the discord then and we can welcome you to the dev team... so we are on track with all stuff

and I'd still need that commit for credits

ankris812, 2022-12-27:
> Good to know where those came from. Only downloaded them from a discord lol

Oh, not all are mine, we just made files for more frequencies that werent available at that time

ankris812, 2022-12-27:
> sure sure. would be nice if you'd join the discord then and we can welcome you to the dev team... so we are on track with all stuff
> 
> and I'd still need that commit for credits

When i find It. A couple of days ago I merged it with the original branch and I don't know where it is now.

ClaraCrazy, 2022-12-27:
lol

closed 2022-12-27 by ClaraCrazy