Flipper-XFW/Xtreme-Firmware #40: Weather Station Error

Cla2000ss, 2022-12-29:
### Describe the bug.

When I try going into Applications/Tools/ Weather Station and try running the application it brings up an Error saying "Load failed Found unsatisfied imports.  I tried flashing the firmware again, and still the Error comes up.  I could easily be doing something wrong since I'm a noob to this, but I tried all I could think of.  Thank you for your time, and I hope it is an easy fix. 

### Reproduction

1. Go into Applications
2. Go into Tools
3. Select Weather Station

ClaraCrazy, 2022-12-29:
Issue Acknowledged. No further information at this point in time. Will update whenever more information exists

jbohack, 2022-12-30:
weather station fixed on latest dev build. will be fixed on release. if you have issues after release feel free to open another github issue up and we'd love to help :D

closed 2022-12-30 by jbohack