Flipper-XFW/Xtreme-Firmware #43: Long press of Power button to power off Flipper

bip3r, 2022-12-29:
### Description of the feature you're suggesting.

I would be great to include in your FW a feature which is present in RM-FW..
When you hold Power button for 2-3s it ask if I want to shut down device.
Right click to confirm.

Nice feature, much easier then strolling through Settings menu to turn it off.
(I wrote the same under Pen Ace YT video).

Nethertheless, great FW Clara!

Another possible feature request - to remove that one unreadable animation.. Others are very nice. And not b/c of the anime girls, but due to its clarity and fluidity (if I can say so.. English is not my native language).

jbohack, 2022-12-29:
thank you so much for the feedback! power off through back button does exist already. it's about five seconds of holding down the power button :) we are also working on changing / adding new animations soon!

bip3r, 2022-12-29:
LOL indeed.. Seems I wasn't patient enough to hold ~6s (maybe 5.5s) :)
So new request - if possible, make it 3s ;)

nameless312, 2022-12-30:
This change has been added.

closed 2022-12-30 by nameless312