Flipper-XFW/Xtreme-Firmware #81: New animations

Toosicc7up, 2023-01-10:
### Describe the bug.

Upon unlocking a new level the new animations unlocked no longer are set as the wallpaper. It usually keeps the same animation or sets one of the older animations.

### Reproduction

Level up

ClaraCrazy, 2023-01-10:
the animations are picked random. every new level gives you an **additional** new one for the system to choose from, they cycle through every X frames (3600 in our case) and all have an equal chance of showing up (besides my personal ones, they have a slightly higher chance lol)

Toosicc7up, 2023-01-10:
Thank you for the clarification. Before when I would unlock a new level it
would automatically put the new animation as the background. Was a nice
touch so you could see the new ones you unlocked now I have ones I've
unlocked I've never even seen three or four levels ago lol

ClaraCrazy, 2023-01-10:
gonna look into if im able to control which animation plays. doesnt seem to be the case tho without a massive rewrite of some OFW code bc its just a pure mess lol. I hate their codebase so much xD

closed 2023-01-10 by ClaraCrazy