Flipper-XFW/Xtreme-Firmware #85: U2F over NFC

TheXHalferX, 2023-01-11:
### Description of the feature you're suggesting.

Hello, Clara) I know, that it is not easy to implement, but is there any chance, that [this](https://github.com/DarkFlippers/unleashed-firmware/issues/206) feature will come to Flippers?

### Anything else?

Thanks for this truly awesome firmware and have a wonderful day/night)

ClaraCrazy, 2023-01-11:
*maybe*. We'll look into it

if it works fine with no/minor modifications, yes. If it needs massive changes, no.

ClaraCrazy, 2023-01-11:
@devsnek Hey, hope you dont mind me dragging you into this.

OFW seems to not want to merge your changes, we'd love to. Tiny issue with your mod is that the normal u2f would no longer work. willing to look into this with our team?

devsnek, 2023-01-11:
@ClaraCrazy The U2F over NFC pull request has not been merged as I'm waiting for their rework of the furi NFC library, which will bring better support for NFC-A and ISO-DEP. Feel free to use the changes as you wish.

ClaraCrazy, 2023-01-11:
> @ClaraCrazy The U2F over NFC pull request has not been merged as I'm waiting for their rework of the furi NFC library, which will bring better support for NFC-A and ISO-DEP. Feel free to use the changes as you wish.

You really expect them to make anything better? Looking at the work they have done in the past few months I'm highly doubtful anything of use will ever be updated to a state that isn't a total clusterfuck. I absolutely feel you, with waiting lol. We'll try to get it running, was just wanting to know if you wanted to help with that

ToXiCRaTt, 2023-01-14:
Oops why have time settings with No clock?... don't really need a clock.

ClaraCrazy, 2023-01-14:
?

legioni87, 2023-01-15:
Clara why am I having issues finding the update file once ive transfered the extracted firmware folder to the flipper. I search in browser then update and then this firmware but I can't seem to find the corresponding update file once there. any suggestions?

closed 2023-01-26 by ClaraCrazy