Flipper-XFW/Xtreme-Firmware #89: slow menus

alexo0o, 2023-01-14:
### Describe the bug.

when you go to passport then back to the main menu and then to the file explorer thing it takes awhile to load

### Reproduction

main menu - passport - main menu - file explorer

jbohack, 2023-01-14:
hello! so file explorer loading times is semi-normal since each time you open the application, it indexes those files in your 'Sub-GHz' folder. the more files you have, the more it needs to load.

jbohack, 2023-01-14:
how long does yours take and how many files do you have in that folder?

alexo0o, 2023-01-14:
it takes about 3 seconds which was very noticeable from the usual instantaneous swich and ive dumped the entire sub ghz folder from uberguidoz's github but i didnt have a problem with that before the update.

jbohack, 2023-01-14:
we changed the default browser on load this fix this issue. not sure why it was loading from users' folders anyhow- let me know if you have issues on the next release sorry for the current inconvenience.

closed 2023-01-14 by jbohack