Flipper-XFW/Xtreme-Firmware #111: NFC Can't emulate hotel keys

RChadwick7, 2023-01-26:
### Describe the bug.

With Unleashed, the Flipper does a dictionary attack, which takes a few minutes. Especially with later builds of Unleashed, this is almost always successful, and it will be able to save and emulate the key. With Xtreme, it reads it very quickly. However, emulating it always fails.

### Reproduction

.

jbohack, 2023-01-26:
sorry for the inconvenience, what type of tag are you trying to emulate? and what are your steps to reproduce this issue?

ClaraCrazy, 2023-02-12:
stale

closed 2023-02-12 by ClaraCrazy

RChadwick7, 2023-02-12:
Sorry, it was my mistake. I'm on the road a lot and scan my key in by habit. I flashed Xtreme at the same time I scanned a key at a new hotel, and it wouldn't emulate. I went back to Unleashed and had the same problem so it's not an Xtreme problem.

jbohack, 2023-02-12:
> Sorry, it was my mistake. I'm on the road a lot and scan my key in by habit. I flashed Xtreme at the same time I scanned a key at a new hotel, and it wouldn't emulate. I went back to Unleashed and [...]

ah yeah sounds like it might not be supported properly yet. nfc for the flipper is still lacking support in many areas. some people -for instance bettse are trying to add more and more functionality for nfc to the flipper though :D