Flipper-XFW/Xtreme-Firmware #119: Remove "Thank You" Animation

DrakkenWolfgang, 2023-01-31:
Hello. New to the Flipper scene. Just got mine a couple of days ago & it has been quite the experience! Have been enjoying tinkering around with it & learning more about it hands on.

I have a slight issue & hoping someone can point me in the right direction.

I have the latest release Xtreme Firmware installed & am curious to know if there is a way to remove "Thank You" splash animation & how would i go about it?

It seems to want to override the stock flipper animation at times. Sometimes a reboot shows the stock flipper animations for a while, then goes back to the "Thank You" animation.

Any info would be appreciated! [image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

Willy-JL, 2023-01-31:
The thank you is the basic fallback builtin animation for this firmware, like for the official firmware its the dolphin looking at the tv on the couch. On flipper the fallback animation is used in 3 cases:
1. No other suitable animation was found. For example you are not high level enough for any of the animations (would occur only on custom packs, the included animations are all scaled correctly with levels).
2. A suitable animation was found, but it failed loading. Could be because of too many frames, some incorrect metadata, or frame files being too big. Again, only an issue on custom asset packs since the pre-included animations were all tested.
3. It gets chosen randomly. When choosing an animation to play its picked randomly between the ones listed in the animation manifest, and the built in animations. So there is a chance that the fallback animation shows up. Depending on how many other animations are suitable, this is more or less likely to happen. If you are level 1, there are 2 possible animations: the level 1 animation (with weight 7), and the fallback animation (with weight 1), so there is a 1 in 8 chance (12%) you see the thankyou animation. If you are level 30, there are 30 suitable animations (all weight 7) plus the fallback one (at weight 1), aka about a 1 in 211 chance  (0.4%) you see the built in animation.

To summarize:
- Dont remove it, it is used as a fallback animation for when things go wrong
- Get more animations so you see it less often

closed 2023-01-31 by Willy-JL

DrakkenWolfgang, 2023-01-31:
> The thank you is the basic fallback builtin animation for this firmware, like for the official firmware its the dolphin looking at the tv on the couch. On flipper the fallback animation is used in 3 cases:
> 1. No other suitable animation was found. For example you are not high level enough for any of the animations (would occur only on custom packs, the included animations are all scaled correctly with levels).
> 2. A suitable animation was found, but it failed loading. Could be because of too many frames, some incorrect metadata, or frame files being too big. Again, only an issue on custom asset packs since the pre-included animations were all tested.
> 3. It gets chosen randomly. When choosing an animation to play its picked randomly between the ones listed in the animation manifest, and the built in animations. So there is a chance that the fallback animation shows up. Depending on how many other animations are suitable, this is more or less likely to happen. If you are level 1, there are 2 possible animations: the level 1 animation (with weight 7), and the fallback animation (with weight 1), so there is a 1 in 8 chance (12%) you see the thankyou animation. If you are level 30, there are 30 suitable animations (all weight 7) plus the fallback one (at weight 1), aka about a 1 in 211 chance  (0.4%) you see the built in animation.
> 
> Also, with v40 we added a check to make sure the same animation is not played twice, so worse comes to worst youre level 1 and you see level 1 anim, then fallback anim, then lvl 1 anim, then fallback anim...
> 
> To summarize:
> - Dont remove it, it is used as a fallback animation for when things go wrong
> - Get more animations so you see it less often

Thanks for the response! Definitely clears some info I didn't understand. My next research point with Flipper will be custom animations/ icons. That is a long read in itself, but I'll figure it out eventually & maybe make some.🤣

Willy-JL, 2023-01-31:
> Thanks for the response! Definitely clears some info I didn't understand. My next research point with Flipper will be custom animations/ icons. That is a long read in itself, but I'll figure it out eventually & maybe make some.rofl

Good luck with that, feel free to join the discord for some help. Also, if you just look online youll see the common and annoying way to do this, which is replacing existing animations and icons on the sd or the firmware. We have our own system with asset packs that takes away a good chuck of that struggle, you can learn about that [here](https://github.com/ClaraCrazy/Flipper-Xtreme/wiki/1.-File-Formats) (the bottom section with animations icons and packs)