Flipper-XFW/Xtreme-Firmware #120: IR sensor barely working

GarytheWorm, 2023-01-31:
### Describe the bug.

IR sensor is very rarely working, I’ve tried in so many times and has only worked twice

### Reproduction

1. Open ir remote
2. Use an ir file

### Target

RI

GarytheWorm, 2023-01-31:
Never mind, my sensor is just being really picky about where i aim it

closed 2023-01-31 by GarytheWorm

ClaraCrazy, 2023-01-31:
Hi there!

This seems like a weird issue, unable to replicate it. See attached video, IR works flawlessly for me, my dev team & around 3k people that have downloaded our Firmware so far, according to GitHub statistics. Whats also important to add, is that our Infrared app is fully stock, besides the Asset change allowing for asset_packs to run, so a Firmware issue can be ruled out.

Theres a few things I can recommend:

1. Check your IR files
2. Make sure that you have a clear view of the target IR receiver
3. Make sure the IR transceiver is cleaned, a dirty front will certainly make things worse
4. Make sure your IR sensor isn't faulty (This can happen, flipper quality control is not the best sadly)
    - To test that, just read a few remotes with it,

Heres a video of it working just fine on two random devices I got:

https://user-images.githubusercontent.com/55334727/215755717-de05cf70-d977-4f83-8077-7c17f56111b5.mp4

ClaraCrazy, 2023-01-31:
guess I was too slow, lmao