Flipper-XFW/Xtreme-Firmware #122: Card Reader/Identifier

Karevski, 2023-02-05:
### Description of the feature you're suggesting.

It would be cool if there is a card identifier/analyser app like the card cloners have. You start it present the card and the app cycles trough the different frequencies and reports when the card has been read. Also it would be cool if after it can drop you in the corresponding app (125kHz or NFC)

SHUR1K-N, 2023-02-06:
Doesn't the "Read" functionality in the RFID/NFC apps already show the chip information as you mentioned by default?

Karevski, 2023-02-07:
> Doesn't the "Read" functionality in the RFID/NFC apps already show the chip information as you mentioned by default?

You have to know if it is 125 or NFC and then is it pico pass etc. i was thinking it would cycle the read modes and notify you when it gets a read

closed 2023-02-07 by ClaraCrazy