Flipper-XFW/Xtreme-Firmware #125: bluetooth freezeing

dontshome, 2023-02-07:
### Describe the bug.

whenever im useing the bluetooth romote it keeps freezeing so i gotta reset my flipper

### Reproduction

connected help the button on mouse and it freezes after a short time

ClaraCrazy, 2023-02-12:
unable to replicate, feel free to open with more infos...

closed 2023-02-12 by ClaraCrazy

dontshome, 2023-02-12:
It's after installing it I connect to use wireless mouse movie it around
some in the same or diff direction it freezes and stops all mouse input
which is the main function I use

Sublimefly, 2023-02-20:
I couldn't recreate this issue, but I did have a lot of trouble getting the bluetooth apps like media remote paired with any of my devices after the last update on Xtreme firmware. For some reason the Flipper Zero isn't showing a name for pairing any longer, instead it displays as unknown. Once I got it paired though the app worked as normal and just shows under the name 'Control' once paired with a PC for example. I've been able to use it to control Plex as well as Netflix still otherwise.

dontshome, 2023-02-20:
I have that pairing issues to but it's being used on  1440p monitor and
it's after so much moving the mouse if I swap keyboard and back it freeze
and continues 2 the firmware for Bluetooth is basically unusable when it
freezes I've been using a diff one now cause of it but wanna use extreme

smannchen, 2023-02-20:
Pretty sure this is a bug they're working on.

dontshome, 2023-02-20:
Idk all I know is it can be frustrating been having these issues with it
for a little while now really want it fixed.