Flipper-XFW/Xtreme-Firmware #128: MORE APPS!!!!

BBoyzthehacker, 2023-02-12:
### Describe the enhancement you're suggesting.

i love this firmware. but... i think it needs some more apps there are many, there is one in dexv that's called ir remote and it helps me a TON and in rougemaster there are alot of apps that i use alot! i was thinking that you could maby just copy and paste in the diff apps from rougemaster also i think that you could add a IR remote dump file or something from

### Anything else?

here is one of the ir dumps u could add
[Flipper-IRDB-main.zip](https://github.com/ClaraCrazy/Flipper-Xtreme/files/10715134/Flipper-IRDB-main.zip)

ClaraCrazy, 2023-02-12:
"More apps"? We have them all, besides the ones that do not function without disabling compiler flags like roguemaster does, which is bad developer practise

ClaraCrazy, 2023-02-12:
And no, we do not just add the IRDB

1. Do you realize how massive that his?
    - It would take ages to flash
    - be a waste of space for everyone that does not need IR

2. We provide firmware, not files for it. We provide a sample size of assets so features can be tested, but other than that this isn't our job.

closed 2023-02-12 by ClaraCrazy

BBoyzthehacker, 2023-02-12:
ok! yeah this helps me understand why you cant (: also im 11 so i dont rlly know what im doing, bye!!