Flipper-XFW/Xtreme-Firmware #136: 467.989 ghz always on

maximed911, 2023-02-15:
### Describe the bug.

467.989 and 468.009 ghz always on when using frequency analyzer in house... it was never detected before update V41_02142023.

unlshd V.30 have made this update:

SubGHz: Remove 467.75 From freq analyzer since it has too much noise (Frequency is still can be used, just excluded from FA to avoid false detections

### Reproduction

just go in frequency analyzer lol

### Target

Frequency analyzer

jbohack, 2023-02-15:
hello, this is actually intended since the frequency analyzer is meant to inform you of all common frequencies that are seen. it isn't noise-related it's just a very common pager frequency that you can read with the 'Pagers' modulation :)

closed 2023-02-15 by jbohack

maximed911, 2023-02-15:
Alright thanks. Sorry for the inconvenience :)

Willy-JL, 2023-02-16:
Small correction: yeah it is background noise, i heard its from the internal chip itself, but not sure. Anyway I would rather keep it and just have users use a higher RSSI threshold instead of removing it completely personally

maximed911, 2023-02-16:
iight np :)