Flipper-XFW/Xtreme-Firmware #137: OPEN

NORMANCOMICS, 2023-02-16:
$ git clone --recursive https://github.com/ClaraCrazy/Flipper-Xtreme.git
$ cd Flipper-Xtreme/

To flash directly to the Flipper (Needs to be connected via USB, qFlipper closed)
$ ./fbt flash_usb

To just compile firmware
$ ./fbt updater_package

If building FAPS:
$ ./fbt fap_dist

If building image assets:
$ ./fbt resources icons dolphin_ext

ClaraCrazy, 2023-02-16:
???

closed 2023-02-16 by ClaraCrazy

NORMANCOMICS, 2023-02-18:
Sorry i honestly have no idea what im doing. Just trying to download yer firmware my fren. Can i request some RARE PEPE animations in the future? I could always help with the art if you were ever interested and the animations as well i would just need to know what size specs you use and what programs.
My stuff is at: normancomics.cent.co

Sorry about the crazy previous posts. Im super n00b to this world. But trying to learn.