Flipper-XFW/Xtreme-Firmware #138: Bad KB not connecting

d4r14h, 2023-02-16:
### Describe the bug.

When I try to connect to the flipper's Bad KB access point, it connected once, then never again. I used an android phone to connect to it.

### Reproduction

Go to BadKB tab
Select payload
Go on phone, to bluetooth devices
Device shows up, error message "Couldn't connect"

### Target

BadKB - Android

ClaraCrazy, 2023-02-16:
documented on the server already
.please check the server before opening issues. (imma edit the known issues thing on gh once out of bed)

Willy-JL, 2023-02-16:
Clara didn't understand the problem here, and this is not answered in the discord. What is answered in the discord is pairing issues with the mobile app, that require a single re-pain once after updating. This is a different issue, with badkb, where after connecting to badkb once it will not connect for future payloads.

This is partially intended, partially not. For the moment you need to unpair the badkb "access point" from the target device each time. **It is intended as a one time use, so that after using badkb on a device, it does not interfere with future payloads by connecting automatically.** We would like however to add some sort of whitelist so that some devices will be allowed to reconnect to badkb automatically. For the time being however, you need to unpair badkb for further connections. I'll reopen the issue so we can track this, and will also mention this behavior either in the FAQ or in the generic guides wiki.

Willy-JL, 2023-02-22:
Small update: we now plan on implementing a config option to allow future reconnects to the BadKB BT access points. Still working on it

whois-hoeless, 2023-03-04:
Done, gonna be added in V42. You will be able to set a toggle to remember the devices or not.

closed 2023-03-04 by whois-hoeless