Flipper-XFW/Xtreme-Firmware #139: subghz bruteforce support for linear garage door openers frequency 318

merrychrishna, 2023-02-16:
### Description of the feature you're suggesting.

subghz bruteforce doesnt work on linear garage door openers frequncy 318 . please can you add this in an update?

ClaraCrazy, 2023-02-16:
Its rolling code

merrychrishna, 2023-02-16:
it is NOT rolling code. i cloned my linear garage door opener 318 and it opens and closes my door every single time. i dont know why you think that. i dont have to jam the signal and i can reuse the same recording over and over.

ClaraCrazy, 2023-02-16:
Guess google lied to me then. Still nothing that can really be "simply added" the way you are hoping for

ClaraCrazy, 2023-02-16:
You'd still need the actual keys for that door, which I don't own. Multiple doors to test, which I do not own. A team to write the logic behind it, which I don't really have. 

Having said that, and simply because brute forcing garage doors is going to land a kid in jail in 0.3 microseconds… For now, the answer is no.

closed 2023-02-16 by ClaraCrazy

merrychrishna, 2023-02-16:
oh ok