Flipper-XFW/Xtreme-Firmware #140: null pointer dereference

Lecker-Duplo, 2023-02-17:
Starting the clock (short press courser left on homescreen) sometimes causes the device to crash/reboot.
Error message:
null pointer dereference

Willy-JL, 2023-02-22:
We had it happen just a couple times but it is very rare and we struggle to reproduce it... we will try to look into it but can't promise anything

ClaraCrazy, 2023-03-09:
@Lecker-Duplo havent seen it happen in a while. we did lots of changes since then to the code. can you confirm the issue is gone?

Lecker-Duplo, 2023-03-09:
I have just updated to the latest version.
Unfortunately, the problem still occurs. Sporadically, but it does occur.
Maybe it has something to do with the 24h D/M/Y metric.
As a test I have set the time and date format to 12h M/D/Y and there the problem does not seem to occur.... at least until now.

m0rd3cai, 2023-03-17:
I can confirm I just put the March 9th release on a clean Flipper and it has done this twice now in 15-20 minutes. Will update if it starts doing it again. It appears to have stopped for now. Will update. This is a new Flipper, not a dev flipper.

Willy-JL, 2023-04-06:
Should now be fixed via 4af2bd01ef6908d2417102e0f7a9fa63e8c142d2, haven't been able to replicate it anymore after that fix. Looks like it was a race condition, the draw callback needed the display brightness, but that was set after the the draw callback, so occasionally it would read the data before being set. Can you please verify if the issue is fixed in the [dev build](https://github.com/ClaraCrazy/Flipper-Xtreme/actions/runs/4624431401)?

Lecker-Duplo, 2023-04-08:
Solved with XFW-0043_28032023

ClaraCrazy, 2023-04-09:
Fixed in latest release

closed 2023-04-09 by ClaraCrazy