Flipper-XFW/Xtreme-Firmware #144: Redmi 11s 5g asking for pass for badBLE

optimus10ro, 2023-02-19:
### Describe the bug.

[image]

When i try to use badBLE on my phone it asks for password but the flipper dose not display any password

### Reproduction

uh, whit a Redmi 11s 5g?

### Target

badBLE

ezpzX1, 2023-02-21:
that happend for me also when i do it on android. on iphone it works fine tho.

Willy-JL, 2023-02-22:
Issue is partly the device's fault. To aid BadKB usage over bluetooth we made it use a different pairing method, that is supposed to only ask for confirmation that pins are equal, **NOT** ask to enter a pin. When this works correctly, the target device just needs to click "Yes", so we made the pairing code not show on flipper. Apparently some devices, particularly some androids and very rarely some linux distibutions / desktop environments, tend to not implement this protocol correctly and ask to enter the pin manually when the protocol should be to just ask for confirmation, but we hid the code on flipper so youre left hanging.

One possible solution would be to have the pairing code show on flipper, but not as prominently as the normal mobile app pairing does. Perhaps show the code in a corner of the BadKB interface... or if possible, force the pin to be a dummy one like 1234 or 0000, if the protocol allows that.

Willy-JL, 2023-02-25:
Fixed via 62dd0aded2f76d69c33578a4695678ebe99a4b8b by showing the PIN code anyway, but in an unobtrusive way. Will be released with V42

closed 2023-02-25 by Willy-JL