Flipper-XFW/Xtreme-Firmware #148: nightstand clock dosent dim backlight

ezpzX1, 2023-02-21:
### Describe the bug.

when i go into the nightstand clock app it doesn't dim the backlight like it dose on RM. it is pretty much useless without that cuz it is very bright when i use it and i dont think that i could sleep with that.

### Reproduction

make it so the backlight auto dims when entering the app

### Target

nightstand clock buug

### Anything else?

i dont know if this is a bug or not but i think it is cause it dose dim on RM

*be nice im 11 years old. and i am new to FAP files so i have no clue what im talking about.*

ClaraCrazy, 2023-02-21:
nightstand clock can be dimmed using up and down keys, just as its intended too. Its functionality is also included in the clock on the main menu.

closed 2023-02-22 by Willy-JL

ezpzX1, 2023-02-22:
thank you! i dident know that you hadto dim it  with the buttions