Flipper-XFW/Xtreme-Firmware #150: Extra Functions for Bluetooth Remote

BoomSec, 2023-03-04:
### Describe the enhancement you're suggesting.

I believe an additional "Clicker" option to be added to the Bluetooth remote could benefit not only the existing users but future users as well. A similar option already exists with the keynote function but I feel users would benefit from an auto clicker function. Suitable for taking multiple photos at once without causing an iPhones burst function to go off. Or repeatedly tapping a button in a game for when that gets too tedious. Even if you allow users to record and replay macros for use in the Bluetooth remote for automatable tasks. I am willing to provide more detail if required

whois-hoeless, 2023-03-04:
Looks doable, we'll look into it.

BoomSec, 2023-03-06:
Thanks legend

TheMany172, 2023-03-06:
Might not be the place to add, would be good to be able to see what device flipper is connected to. Be able to disconnect from it etc

TheMany172, 2023-03-06:
Just checked and I actually cannot get it to connect or show up on another devices Bluetooth scan oddly

I don't believe it is connected to anything, as when loading the keyboard, it says waiting for connection

BoomSec, 2023-03-07:
Raise an issue ticket then @TheMany172

BoomSec, 2023-03-07:
Extra points if you are able to make swiping possible. Imagine all the people using the flipper to swipe on apps like tinder 😜