Flipper-XFW/Xtreme-Firmware #152: First letter capitalization

ms4sman, 2023-03-06:
### Describe the bug.

I tested this in a few different spots around the firmware just now to make sure it's not just one, and so far it seems to happen everywhere I tried.

When the on screen keyboard comes up, for example, to name something like a manually entered record, it automatically wants to capitalize the first letter. This is fine and I have no issue with it. But it seems there's no way to NOT capitalize the first letter. You'd expect that holding the OK button on the letter would give a lowercase letter, similar to how normally it would give an uppercase, but no matter what, it seems to always want to put a capital for the first letter.

On other custom firmwares, I seemed to encounter this issue sporadically and sometimes it behaved differently in different apps but so far, on XFW it seems to be consistent.

Kind of a small nitpicky thing, but it is frustrating as I'd sometimes like a lowercase letter for the first character I type. And since (as far as I know) there's not a way to move the cursor without just backspacing, you can't go back and enter a lowercase AFTER putting the first letter or something.

Thanks for all the hard work you guys all do!

### Reproduction

Go into an app (I just tested it with the Name Changer, as well as Add Manually for a SubGhz item, and Add Manually for an iButton item, but I think it happens everywhere)

When the on-screen keyboard comes up to enter a name, try to enter a lowercase letter for the first one. The keyboard shows all caps, and so unsurprisingly clicking a letter puts a capital. But holding the OK button on the letter does not give a lowercase either. Nor does putting a letter, then backspacing to remove it, and trying again (this used to help on some other custom firmwares I used).

Willy-JL, 2023-03-06:
Fixed in dev branch., will be in v42

closed 2023-03-06 by Willy-JL

ms4sman, 2023-03-06:
Heck yeah! Thanks!