Flipper-XFW/Xtreme-Firmware #153: Bluetooth remote not broadcasting/connecting

TheMany172, 2023-03-07:
Noticed that I cannot use any of the Bluetooth remote features

- load app from applications or browser
- can open any of the internal apps
- light on flipper doesn't flash to indicate connecting, it also doesn't show up on phone or laptop's Bluetooth scan. So cannot connect

I thought maybe it was just connecting to something else, but when I open the keyboard internal app, it says 'waiting for connection'

Tried moving away from other stuff with just my phone and flipper, same result

If possible would also be good to add one or more of these 
- [ ] able to see device label that the Bluetooth remote is connected to
- [ ] be able to disconnect from is using the flipper
- [ ] be able to browse and choose what to connect to from the flipper

Thanks and good luck

bHub, 2023-03-08:
Just tested with original firmware, when you open the bluetooth keyboard app a device named "Control <yourflippername>" will show up if you do a scan, nothing happens with Xtreme fw, so there's something messed up with BT, I guess.

TheMany172, 2023-03-09:
Yeah it was the same on roguemaster, nothing on this one tho

Willy-JL, 2023-03-09:
Fixed in v42. Please be patient just a little longer

closed 2023-03-09 by Willy-JL