Flipper-XFW/Xtreme-Firmware #154: BadUSB missing after firmware upgrade

deebee-nl, 2023-03-08:
### Describe the bug.

After updating to XFW-0041_02142023 the BadUSB option is missing the flipper. Not sure if it's me doing something wrong but removing apps and re-installing FW does not restore the option.

Is it me? Or have others reported this as well?

### Reproduction

[image]

[image]

It's completely footsie and not sure how to fix unless moving to another FW which is not the desirable option :-)

### Logs

```Text
612 [APP] qFlipper version 1.2.2 commit ba67025d 2022-12-08T13:31:11
613 [APP] OS info: Windows 10 Version 2009 10 10.0.19045 Qt 5.15.2
796 [DBG] Trying serial port FLIP_DEEBEE at \\.\COM8
796 [DBG] Using  serial port FLIP_DEEBEE at \\.\COM8
796 [UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
797 [UPD] Fetched update information from https://update.flipperzero.one/qFlipper/directory.json
808 [RPC] Starting RPC session...
845 [APP] Another instance was prevented from running
874 [RPC] RPC session started successfully.
874 [RPC] (1) System Protobuf Version START
936 [DBG] Detected protobuf version: 0.14
936 [RPC] (1) System Protobuf Version SUCCESS
936 [RPC] (2) Property Get START
1057 [RPC] (2) Property Get SUCCESS
1057 [RPC] (3) Storage Info @/ext START
1114 [RPC] (3) Storage Info @/ext SUCCESS
1114 [RPC] (4) Storage Stat @/ext/Manifest START
1176 [RPC] (4) Storage Stat @/ext/Manifest SUCCESS
1197 [RPC] (5) System Get DateTime START
1254 [DBG] Flipper time skew is -993 milliseconds
1254 [RPC] (5) System Get DateTime SUCCESS
1274 [RPC] (6) System Set DateTime START
1334 [RPC] (6) System Set DateTime SUCCESS
1336 [RPC] Stopping RPC session...
1337 [RPC] Attempting to unload protobuf plugin...
1337 [RPC] Unloaded protobuf plugin.
1337 [RPC] RPC session stopped successfully.
1337 [REG] Registering the device
1337 [DEV] Version: 0.74.3 commit: XFW-0041 radio: 1.13.3
1337 [BKD] Current device changed to "DeeBee"
1456 [RPC] Starting RPC session...
1614 [RPC] RPC session started successfully.
1614 [RPC] (1) Gui Start ScreenStream START
1674 [RPC] (1) Gui Start ScreenStream SUCCESS
69033 [RPC] Serial connection was lost.
69033 [RPC] Stopping RPC session...
69033 [RPC] Attempting to unload protobuf plugin...
69033 [RPC] Unloaded protobuf plugin.
69033 [RPC] RPC session stopped successfully.
69063 [REG] Device disconnected: VID_0x483:PID_0x5740
69063 [BKD] Last device was disconnected
69845 [REG] Detected new device: VID_0x483:PID_0x5740
69872 [DBG] Trying serial port FLIP_DUR4 at \\.\COM4
69873 [DBG] Using  serial port FLIP_DUR4 at \\.\COM4
69873 [RPC] Starting RPC session...
70013 [RPC] RPC session started successfully.
70013 [RPC] (1) System Protobuf Version START
70373 [DBG] Detected protobuf version: 0.14
70373 [RPC] (1) System Protobuf Version SUCCESS
70373 [RPC] (2) Property Get START
70593 [RPC] (2) Property Get SUCCESS
70613 [RPC] (3) Storage Info @/ext START
70696 [RPC] (3) Storage Info @/ext SUCCESS
70713 [RPC] (4) Storage Stat @/ext/Manifest START
70793 [RPC] (4) Storage Stat @/ext/Manifest SUCCESS
70813 [RPC] (5) System Get DateTime START
70893 [DBG] Flipper time skew is -1632 milliseconds
70893 [RPC] (5) System Get DateTime SUCCESS
70913 [RPC] (6) System Set DateTime START
70973 [RPC] (6) System Set DateTime SUCCESS
70993 [RPC] Stopping RPC session...
70995 [RPC] Attempting to unload protobuf plugin...
70995 [RPC] Unloaded protobuf plugin.
70995 [RPC] RPC session stopped successfully.
70995 [REG] Registering the device
70995 [DEV] Version: 0.74.3 commit: XFW-0041 radio: 1.13.3
70995 [BKD] Current device changed to "DeeBee"
71113 [RPC] Starting RPC session...
71258 [RPC] RPC session started successfully.
71258 [RPC] (1) Gui Start ScreenStream START
71333 [RPC] (1) Gui Start ScreenStream SUCCESS
```

Willy-JL, 2023-03-08:
With V41 we renamed BadUSB to BadKB (bad keyboard) since it now also includes a bluetooth mode

closed 2023-03-08 by Willy-JL

deebee-nl, 2023-03-08:
@Willy-JL - Thanks! It was me and my ignorance.