Flipper-XFW/Xtreme-Firmware #157: Standalone MFKey32 app (.fap)

WatskeBart, 2023-03-09:
### Describe the enhancement you're suggesting.

Could you add [FlipperMfkey](https://github.com/noproto/FlipperMfkey) to XFW?
Current API version is not compatible with XFW.

This app (.fap) makes Mifare cracking possible without using the Flipper companion app on your phone.

### Anything else?

Keep up the clean coding, and hope v42 will be available soon ;)

skynet01, 2023-03-10:
Meanwhile you can the latest fap file here: https://github.com/xMasterX/unleashed-extra-pack/tree/main

ClaraCrazy, 2023-03-10:
a fap literally doesnt help at all, since it cant be used on different fw

skynet01, 2023-03-11:
I thought fap is API dependent? So as long as both Unleashed and your firmware is on the same version wouldn't they work?  I used a few faps from that link including the one that's requested and they all worked for me.

ClaraCrazy, 2023-03-12:
Updated previous comment, not sure what I was on about. 

FAPs work as long as our API version is the same, yes.

Willy-JL, 2023-03-20:
Added, will be in v43

closed 2023-03-20 by Willy-JL

LevitatingBusinessMan, 2023-04-02:
Does this app add anything seeing as MFKey32 seems to be built into the firmware?