Flipper-XFW/Xtreme-Firmware #159: Bluetooth remote enhancements - based on previous issue raised

TheMany172, 2023-03-09:
### Describe the enhancement you're suggesting.

If possible would also be good to add one or more of these

- [ ] able to see device label that the Bluetooth remote is connected to

be able to disconnect from is using the flipper
be able to browse and choose what to connect to from the flipper

ClaraCrazy, 2023-03-09:
- done
- not possible
- turn off bt
- not possible

closed 2023-03-09 by ClaraCrazy

Willy-JL, 2023-03-09:
For more info on the not possible's: flipper uses the BLE light stack, which only allows acting as a 'slave' and not a 'master' devices (yes the technical terms could be better). This means that devices connect TO flipper, while flipper cannot see or connect to other devices

TheMany172, 2023-03-09:
Thanks both for the elaboration, explaination and speed - I will bear it in mind

I have tried connecting to my android phone and the scan shows a mac address - I have tried connecting to a macbook and it shows 'generic device'

once either are actually connected it then shows 'Control [nameofflipper]'

not the end of the world tho