Flipper-XFW/Xtreme-Firmware #162: Crash - semi-infrequent when selecting bad KB

TheMany172, 2023-03-09:
### Describe the bug.

Description - when selecting bad KB from the menu, get a crash

[image]

Could well be other instances other than selecting bad KB, have been playing with it since last release (hour or so) and i have seen this screen probably about 8 times during that (admitedly high) use

### Reproduction

1. switch on
2. press middle button
3. press up 4 times
4. (on bad KB) press middle button
5. Crash happens usually before it loads the files inside - sometimes a short delay afterwards

Willy-JL, 2023-03-09:
That crash is typically from a too low value for the app stack size... understandable considering we added functionality to it, but never seen that happen before in testing... Easy enough fix tho

TheMany172, 2023-03-09:
i was just trying to replicate with more precision - and whilst it is not a clear set of steps - it seems if you go through the menu and open each app in turn

and then you go back and follow steps 2+ it is quite easy to get it to replicate

so new steps would be...

1. switch on
2. middle button
3. open first app with middle button
4. press back and then down
5. middle button again to open next app in list
6. repeat until you have gone through the whole list
7. press back multiple times to get back out of whatever you are doing
8. THEN - press middle button, followed by up 4 times (to bad KB) and then press middle button

I just did this and it did it 3/4 times ish

Willy-JL, 2023-03-09:
Still cant manage to replicate this... Are you in out Discord by chance? would be easier to debug if you created a ticket for this

TheMany172, 2023-03-09:
i have just joined - where is best to post?

Willy-JL, 2023-03-10:
Fixed in dev branch via 657f1258f718d861bf776cfeaec4af6b670e9a87, also updated the [release assets for V42](https://github.com/ClaraCrazy/Flipper-Xtreme/releases/tag/XFW-0042_09032023)

closed 2023-03-10 by Willy-JL