Flipper-XFW/Xtreme-Firmware #163: Bad KB doesn't work

slimshotty, 2023-03-09:
### Describe the bug.

Bad KB doesn't work. After trying to access, no other options work except for Settings. Soft reboot corrects the issue until you try to access Bad KB again.

### Reproduction

1. Switch on
2. Access Bad KB.
3. Access anything else but Settings as Settings will work.
4. To recover, soft reboot device.

Willy-JL, 2023-03-09:
Cant reproduce... mind sending a video of it happening, or joining our discord and making a ticket for this?

slimshotty, 2023-03-09:
Now it is working for me, however, after repeating the steps, I received the Oh shit screen stating there was some type of overflow.  I should have captured the screen since I hit it twice.  As I continued to try to access Bad KB, it finally worked.

I'm wondering if it's because of the amount of scripts/nested folders I have under Bad KB.

I promise next time to capture a video or at the very least a screenshot.

closed 2023-03-09 by slimshotty

slimshotty, 2023-03-09:
[image]

Willy-JL, 2023-03-09:
So this is a duplicate of #162 then. Good to know it’s not 2 separate issues