Flipper-XFW/Xtreme-Firmware #165: Barcode app save display

salutcalin2002, 2023-03-10:
### Description of the feature you're suggesting.

I have the barcode dap app from another fw and works amazing to generate and save barcodes as well and display them . It's not included yet would be fun to have it for all users . Directly from the firmware.

Willy-JL, 2023-03-10:
it is included, look in misc

salutcalin2002, 2023-03-10:
It's not the same one this let's you as well to save and edit your own and do a list there two of them .

salutcalin2002, 2023-03-10:
It's called barcode.fap not barcode generator .there different

Willy-JL, 2023-03-10:
mind providing a link then? this is really generic...

salutcalin2002, 2023-03-10:
Yes if I could uploaded somewhere immediately retuning with it on weshare

salutcalin2002, 2023-03-10:
[https://wetransfer.com/downloads/e36ead9bba3bf01fa9ca71d4b804e59b20230310105836/e595b4e485274378f16df2e1494fcfb420230310105847/3f819c](https://wetransfer.com/downloads/e36ead9bba3bf01fa9ca71d4b804e59b20230310105836/e595b4e485274378f16df2e1494fcfb420230310105847/3f819c?trk=TRN_TDL_01&utm_campaign=TRN_TDL_01&utm_medium=email&utm_source=sendgrid)

there are the two one in the fw and the one missing :) 

thank you so much !!! best fw ever ! i told everyone on reddit and git :) how professional it is

salutcalin2002, 2023-03-10:
https://we.tl/t-NDX9J1i2ui?utm_campaign=TRN_TDL_05&utm_source=sendgrid&utm_medium=email&trk=TRN_TDL_05

ClaraCrazy, 2023-03-10:
not helping at all

ClaraCrazy, 2023-03-10:
either send us the source or we cant include it

salutcalin2002, 2023-03-10:
this is the the source
https://github.com/Kingal1337/flipper-barcode-generator

Willy-JL, 2023-03-17:
Done, will be in v43

closed 2023-03-17 by Willy-JL