Flipper-XFW/Xtreme-Firmware #168: On screen Keyboard Issue

BrunoKlu, 2023-03-11:
### Describe the bug.

When navigating between numbers of the numeric keyboard on the right of the letters keyboard : 

If 8 is selected , then pressing up arrow, 6 is selected instead of 5 and vice versa

if 7 is selected, then pressing up arrow, 5 is selected instead of 4 and vice versa 

if 9 is selected, then pressing up arrow, selection disappears.

if “save” is selected, then pressing up arrow goes to 4 instead of “delete” and vice versa

Hope my explanations are more or less clear…

Thanks 🙏 

### Reproduction

1. Go anywhere in Flipper where you use the on screen keyboard
2. On the keyboard navigate to the numeric keyboard with arrows 
3. Put selection on “save” “7”,”8” or “9”
4. Press up arrow

Willy-JL, 2023-03-11:
Right, added to todo list

Willy-JL, 2023-03-14:
Fixed via 2b68bf7542d8d54b5cb2f3e75e0a8062cf75f2a7

commit 2b68bf75 (2023-03-14): Update keyboard offset logic (fix #168)
closed 2023-03-14 by Willy-JL