Flipper-XFW/Xtreme-Firmware #169: Bluetooth remote - Part 2

bHub, 2023-03-11:
Referring to issue  "[Bluetooth remote not broadcasting/connecting #153](https://github.com/ClaraCrazy/Flipper-Xtreme/issues/153)" closed as fixed in version 42, I want to ask if the functioning is correct:
- first I need to open BadKB app and set the connection mode to Bluetooth
- then I need to save the BT device name and MAC address, with default or modified parameters
- finally I could start the Bluetooth remote app and pair the device

If I didn't follow these steps I'm unable to see the "FlipperName Control" device to pair .

Willy-JL, 2023-03-11:
cant replicate, works fine here.... both with bt on and off in badkb settings, the bluetooth remote works fine... can you join the discord and open a ticket, or show a video of this happening?

bHub, 2023-03-13:
Today it seems to work as expected.. I'm going to test further and see if I can reproduce the problem.