Flipper-XFW/Xtreme-Firmware #170: brute force rfid and nfc

KLK19, 2023-03-12:
### Describe the bug.

it seems since updating to latest version of firmware it deleted all my saved files for my codes ??? can anyone help or do i have to go in a reload them all again. 

### Reproduction

when trying to brute force NFC or RFID

whois-hoeless, 2023-03-12:
What files exactly. When you talk about bruteforcing I suppose you mean the files being used by RFID? Can you mind giving more info on your problem and showing us some images or joining our discord and opening an issue?

ClaraCrazy, 2023-03-12:
Ignore that stupid comment. Really tired of this man at this point. 

Deleted files are not something the Firmware can do. The only files touched by *any* Firmware, are the internal Assets, and animations within the Dolphin (& in our case Dolphin-custom) folder(s). 

I use RFID on a daily basis, as otherwise I'd be locked out of my apartment (f*ck keys) so I can confirm that files are **not** being deleted.

When it comes to NFC files it is the exact same. I dont use NFC on a daily basis, however I still can confirm that files I've had for weeks are still here, outside of the default ones we ship. 

I suggest you double-check using like qFlipper, I tend to miss a file at times, maybe thats what happened. Also feel free to join our Discord for a proper "live" support if needed. For now, I will be marking this issue as faulty, as it is not replicable, the firmware does not touch such files, and there is not a single other case like it.

KLK19, 2023-03-12:
thank you i will do that also i will join the discord much appreciated

KLK19, 2023-03-12:
BTW your animations are super dope can't lie lol ...best firmware by far for the flipper

closed 2023-03-12 by KLK19