Flipper-XFW/Xtreme-Firmware #172: POCSAG Pager text overlap

Brereid98, 2023-03-15:
First of all I love this firmware, easily the best I've tried, not 100% sure if this is an issue for you but I thought I'd let you know because the POCSAG pager app is included with your firmware, there is a formatting issue with the reader meaning that the actual message is not able to be read, it seems to happen regardless of message length too.
[image]

ClaraCrazy, 2023-03-22:
idk how this issue went under the radar for a week. We'll look into it and fix it for the next release

Willy-JL, 2023-03-24:
So, seems to me like the issue is happening here:
https://github.com/ClaraCrazy/Flipper-Xtreme/blob/44a1eb77517e33d39bb208015e83467b02436ae5/applications/plugins/pocsag_pager/protocols/pocsag.c#L157-L174

This function is called multiple times, but the `result_ric` string is never reset in here, it just has the same text appended to it multiple times. I might be mistaken, but i feel like `result_ric` is supposed to be set once, while `result_msg` is what is supposed to have text appended to it... With 24d42a46aed96ded48b7eb5e9c463b4e446a6b01 I have made it so it resets the string at the beginning of `pocsag_message_done()`, thats what makes the most sense to me.

However, I have no way of testing / verifying this... can you please try this dev / beta build and test if the issue is fixed, or if it causes more problems?

[f7-update-XFW-24d42a4.zip](https://github.com/ClaraCrazy/Flipper-Xtreme/files/11067165/f7-update-XFW-24d42a4.zip)

Brereid98, 2023-03-24:
Perfect, my transmitter is currently at home but I'll set it up and should be able to let you know in a couple of hours

Brereid98, 2023-03-25:
Looks to be working perfectly now, new update looks amazing too by the way

Willy-JL, 2023-03-25:
Great, thanks <3

closed 2023-03-25 by Willy-JL