Flipper-XFW/Xtreme-Firmware #177: BadUSB BT Connection requires app restart

Kuzuru, 2023-03-21:
### Describe the bug.

At the moment, working with BadUSB via Bluetooth works with a small bug: When connecting to the flipper via Bluetooth BadUSB APP does not immediately detect the device as expected. You have to re-enter badusb after connecting

### Reproduction

1. Open BadUSB (for example `demo_windows`)
2. `< Config`
3. Select `Connection: BT`
4. Connect to your flipper
5. Approve code
6. ?????
7. GOT: No reaction on the connection
8. Closing `demo_windows`
9. Opening it again and voila! It works

### Anything else?

I recorded it in video just in case it could help: https://youtu.be/uEDYI_rPSa8

Willy-JL, 2023-03-21:
Works fine here... are you sure you installed v42 after the hotfix?

Kuzuru, 2023-03-21:
Yep I'm sure. I'd installed v42 from `main` branch from releases tab

Kuzuru, 2023-03-21:
FW: `XFW-0042_09032023`

Willy-JL, 2023-03-21:
yeah... exactly... v42 had issues at launch, we fixed them a little after, files are updated, but its not a new release. i suggest you reinstall

Kuzuru, 2023-03-21:
I guess I can go on and install dev...?

Kuzuru, 2023-03-21:
Reinstalled FW using this one:
[image]

Unfortunately, reinstalling didn't fix my problem

closed 2023-04-05 by Willy-JL