Flipper-XFW/Xtreme-Firmware #181: Bluetooth remote

GreenGoth, 2023-03-23:
### Describe the bug.

Hi. Installed your firmware - Bluetooth remote stopped working. Reflashing, deleting from pairing memory - had no result.

Thank you in advance, you do very cool stuff!

### Reproduction

Can not connect to the phone and control, say, a music player

ClaraCrazy, 2023-03-26:
Cant replicate. Works fine on tested devices of staff. Please reply with additional info, firmware version, android / iOS version, if possible a recording of the issue etc. the more the better.

GIACOMO-JPG, 2023-03-31:
I have the same but only nfc remote emulate works on the app.
iPhone 12 16.4 
XFW-0043_28032023

[https://imgur.com/a/izlipPI](https://imgur.com/a/izlipPI) 

pressing on emulate on everything except nfc it start but can’t be stopped and doesn’t work and doesn’t appear on the FZ

ClaraCrazy, 2023-03-31:
> I have the same bug but only nfc remote emulate works on the app.
> iPhone 12 16.4 
> XFW-0043_28032023
> 
> [https://imgur.com/a/izlipPI](https://imgur.com/a/izlipPI) 
> 
> pressing on emulate on everything except nfc it start but can’t be stopped and doesn’t work and doesn’t appear on the FZ

completely different issue... and it doesnt work bc the app is broken on cfw

wasabicoat, 2023-03-31:
I have the same issue when I try to use a remote control (e.g., a TikTok controller). When I connect it to my iPhone, it works fine, but when I try to connect it to my Android device (Samsung Z Fold4 with Android 13 and One UI 5.1), it doesn't find Bluetooth to connect. I have already unpaired my previously connected Bluetooth controller and Bluetooth connection for the Flipper app, but it still doesn't find the device for Android. However, for iOS, it keeps finding it. I tried installing the Rough Master firmware, and it found Bluetooth and was able to connect.

Rwmtz, 2023-04-05:
I have the same issue with connecting for bt remote  when going to keyboard it says waiting for connection. And the media is not connected also.

Codeplaya, 2023-04-07:
> completely different issue... and it doesnt work bc the app is broken on cfw

Agreed, emulate does not work with any cfw, EXCEPT V42. It really worked with V42 and V42 only. That's why it gets reported so much.

ClaraCrazy, 2023-04-08:
open a new issue for your complaints. This is a nightmare to handle.

charlieporth1, 2023-04-08:
> I have the same issue when I try to use a remote control (e.g., a TikTok controller). When I connect it to my iPhone, it works fine, but when I try to connect it to my Android device (Samsung Z Fold4 with Android 13 and One UI 5.1), it doesn't find Bluetooth to connect. I have already unpaired my previously connected Bluetooth controller and Bluetooth connection for the Flipper app, but it still doesn't find the device for Android. However, for iOS, it keeps finding it. It's worth noting that I'm using Flipper firmware version XFW0043.

It's funny say that because I have that on a Galaxy S22 Ultra, One UI 5.1 Android 13

Willy-JL, 2023-04-09:
This issue turned into  amess. As Clara said make your own issues, don't hijack existing unrelated issues.

As for the bt remote not connecting due to the device not showing up, thats now fixed in dev branch and will be in next release

closed 2023-04-09 by Willy-JL