Flipper-XFW/Xtreme-Firmware #182: USB Mass Storage & SubGhz SOS

HyperMuffin12, 2023-03-26:
### Description of the feature you're suggesting.

Although being slow, I believe an app for USB Mass Storage would be useful for accessing files from the flipper more conveniently.

Please see https://github.com/flipperdevices/flipperzero-firmware/pull/1060 to get an idea of what I mean.

I also feel an app for Subghz Morse Code SOS could be useful, or an app for inputting text and having it transmitted as Morse Code.

https://github.com/arha/dolphin-sos is similar to what I'm suggesting but crashes on any button press.

ClaraCrazy, 2023-03-26:
The issue with mass-storage is that its not working properly as of yet. If you have a look on the projects tab, you will see that its something we are already working on. 

as for morse code... I dont see why, as we already have a proper subghz chat, why complicate it with morse?

HyperMuffin12, 2023-03-26:
I was thinking more so it can be picked up and decoded by someone with a walkie talkie or ham radio.

Might be impractical I'm not sure, but I felt it was worth a suggestion.

ClaraCrazy, 2023-03-26:
:cathmm: valid

TheXHalferX, 2023-03-29:
By the way, this idea of emulating Mass Storage form ISO files. The fact, that speed is capped at 500 kb/s is not good, but sometimes (i.e. I have only one computer) not having already flashed drive can be a disaster. I'm trying to investigate code from this [https://github.com/kevinwallace/flipperzero-firmware/tree/mass_storage](url) commit and it seems that it's VERY outdated, but working properly in build (0.50 or smth). Even loading small ISO of minimal Linux can be 15 -30 minutes long. Maybe it is still worth a shot? Anyway, have a great day/night/other, you beautiful devs)

Zarcolio, 2023-06-07:
Because of the lack of a mass storage function, I wrote a simple Ducky script that launches an executable from a removable drive you hook-up together with a Flipper. Have look at https://github.com/Zarcolio/Duckyscript/blob/main/Ducky%20script/Offensive/ExeFromUsb.txt

[image]

Willy-JL, 2023-07-26:
mass storage is now working and will be released shortly, sos should go into its own issue.

HyperMuffin12, 2023-07-27:
Is there any documentation on how to use the app? 

I have put some files in /ext/apps_data/mass_storage/ 

Despite this none of these files appear in that directory when I open the app.

Any help would be appreciated

TheXHalferX, 2023-07-27:
You should use *.img files to appear in the menu, tested, works great

PR #340 (2023-07-27): V50 Release
### Added:
- SubGhz: Support for new amp module (flux capacitor from rabbitlabs)
- Storage: New setting to wipe device (both internal and external)
- Desktop: Add keybind option for wipe device, add button to reset keybinds
- CLI: New `input keyboard` command for controlling flipper with computer keyboard (typing into textboxes too)
- BadKB: Added French Canadian layout (by @francis2054)
- Furi: Added boot status screens, with led indicator and logging too
- Furi: Clear error message for "out of memory"
- GUI: New keyboard API for custom additional symbols
- GUI: New variable item  [... cut]
closed 2023-07-27 by Willy-JL