Flipper-XFW/Xtreme-Firmware #189: Emulating RFID or sending RFID from mobile app broken again

Codeplaya, 2023-03-31:
### Describe the bug.

On V42 this was working.

Currently Whwn you open mobile app, go to the archive tab, choose any saved items, marked as RFID 125 or  Sub-GHz then pressing Send for the Sub-GHz or Emulate for the RFID 125, does not activate anything on flipper.

NFC and iButton works, however

### Reproduction

Read and save any RFID 125 to Flipper.
Download and install Flipper mobile app.
Open app, pair your flipper to your phone via bluetooth.
Wait until app synchronizes your saved items to the archive.
Open archive tab in the mobile app(middle, folder icon on the bottom ribbon)
Tap your saved RFID125 on the mobile app
Tap Emulate
Nothing happens.
Read eand save any NFC to Flipper.
Open archive tab in the mobile app(middle, folder icon on the bottom ribbon)
Tap your saved NFC on the mobile app
Tap Emulate
Flipper starts emulating.
Tap Emulate again
Flipper stops emulating.

ClaraCrazy, 2023-03-31:
nothing changed on that code, the mobile app is known to not work with cfw

closed 2023-03-31 by ClaraCrazy

Codeplaya, 2023-04-07:
While you can make that argument, I mostly agree. It has never worked with cfw. But it was working with V42. Broken with V41. So it must be related to some other part of the code.