Flipper-XFW/Xtreme-Firmware #190: Screen error graphics

park0urakos, 2023-04-01:
### Describe the bug.

screen error start  with "ligma" flipper screen restart, almost every animation change follow with error, also flipper zero getting slower. 

### Reproduction

 restoring flipper zero with ofw have no issius, any other action not help.

salutcalin2002, 2023-04-01:
Same issue some weird text over other text Ligma and then flipper crash reboot

wfosborn, 2023-04-01:
Same issue here after locking keypad and leaving it overnight. Somewhere along trying to reboot and get it working threw a null pointer  dereference. Had to restore original firmware, haven't been able to put Xtreme back on it yet without the same thing happening though.

achtnullzwei, 2023-04-01:
Same issue. Going to try wfosborn's suggestion on formatting the card to fix this.

park0urakos, 2023-04-01:
I format sd card and restore flipper zero, I install previous version of Xtreme and im ok for the moment till next update.

salutcalin2002, 2023-04-01:
[image]

salutcalin2002, 2023-04-01:
[image]
[image]
[image]

maddawg836, 2023-04-01:
Same here. Keep getting ligma almost every other screen sometimes. I can go back to unleashed again and all animations work again.

ClaraCrazy, 2023-04-01:
@maddawg836 @achtnullzwei @wfosborn @salutcalin2002 @park0urakos 

Welcome to the world of april fools

get some help you basement dwellers with no sense of reality

achtnullzwei, 2023-04-01:
@ClaraCrazy good one.

maddawg836, 2023-04-02:
You know I knew it was a screen animation since the real time clock was fine and I didn't get the issue yesterday which is the first time installing this cfw. That was a good one, ligma.... 

> @maddawg836 @achtnullzwei @wfosborn @salutcalin2002 @park0urakos
> 
> Welcome to the world of april fools
> 
> Yall are goofy
> 
> [image]

You know I knew it was a screen animation since the real time clock was fine and I didn't get the issue yesterday which is the first time installing this cfw. That was a good one and ironic timing, ligma....

wfosborn, 2023-04-02:
@ClaraCrazy My family got a good laugh out of it later this morning when I finally pieced it together too. 😂 Even told myself this morning "Remember it's April Fool's and don't fall for anything this year." 🤦🏻‍♂️

salutcalin2002, 2023-04-02:
Amazing team xtreme amazing firmware 🍻💪😃

jahmaogak0, 2023-04-02:
L3_Sunflower_128x64 is the animation name btw

closed 2023-04-08 by Willy-JL