Flipper-XFW/Xtreme-Firmware #192: Lock screen has no animations

synicalkid, 2023-04-06:
### Describe the bug.

With the latest release, the lock screen has lost all animations. I worry about screen burn. Right now, when I lock,  I get the default Dolphin with ability only to add or remove displays from the lock screen. The prior release, in the lock screen, still showed animations from the current pack (eg. watchdogs.)

### Reproduction

1. Lock the flipper
2. Notice no animations, just a single display.
3. Be sad there might be screen burn.

aidan9898, 2023-04-07:
There's also no way to lock and shut off the device at once like there used to be:/ there definitely needs to be some kind of animation and the ability to lock with PIN code and shut down

Willy-JL, 2023-04-08:
Screen burn is not an issue with flipper's lcd, don't worry about that. And it's intended for the lock screen to be static. As for the option for lock + turn off, we're planning to add that back

ClaraCrazy, 2023-04-09:
Fixed in latest release

closed 2023-04-09 by ClaraCrazy