Flipper-XFW/Xtreme-Firmware #195: NFC Generator / Functions

BoomSec, 2023-04-08:
### Describe the enhancement you're suggesting.

Allow users to generate nfc codes based on what information they want to share. For example, allow users to share files stored on the flipper making for easy transfers during flipper meetups. Or allow people to enter urls into the flipper and then share them via nfc. A possible use case would be an nfc business card / contact. Tapping a phone against the flipper would allow users to add the contact that is being stored in the nfc code.

park0urakos, 2023-04-08:
I tried to pay with NFC that i copied my Bank account and emulated, but every pos system that i tried not recognise it. Any suggestions - bug - recommendation - issue - help?

ClaraCrazy, 2023-04-08:
> I tried to pay with NFC that i copied my Bank account and emulated, but every pos system that i tried not recognise it. Any suggestions - bug - recommendation - issue - help?

Its not possible to pay with a flipper, lol

ClaraCrazy, 2023-04-08:
> ### Describe the enhancement you're suggesting.
> Allow users to generate nfc codes based on what information they want to share. For example, allow users to share files stored on the flipper making [...]

You can already do that on a phone and transfer it to the flipper. creating a whole app is tedious and the added value of this just isnt enough to justify it right now. 

-> Backlog'd

park0urakos, 2023-04-08:
> > I tried to pay with NFC that i copied my Bank account and emulated, but every pos system that i tried not recognise it. Any suggestions - bug - recommendation - issue - help?
> 
> Its not possible to pay with a flipper, lol

Thnx for help.

Willy-JL, 2023-06-18:
Now fully implemented, will be in next release. Not on github yet however, not gonna put it anywhere until we release so a certain someone can't steal it (I know you're reading this, hey RM! :wave: )

Preview:
[image]

BoomSec, 2023-06-19:
Thanks for bringing this back you legend 😎😎

analyserdmz, 2023-06-19:
> Now fully implemented, will be in next release. Not on github yet however, not gonna put it anywhere until we release so a certain someone can't steal it (I know you're reading this, hey RM! 👋 )
> 
> Preview: [image]

Any potential release date for this? 💯

Willy-JL, 2023-06-19:
> Any potential release date for this?

[image]

BoomSec, 2023-06-20:
Fr asking how long a piece of string is

r4p7us, 2023-06-29:
I did something like this..

[Termux_open_by_Raptus.zip](https://github.com/ClaraCrazy/Flipper-Xtreme/files/11905140/Termux_open_by_Raptus.zip)

ClaraCrazy, 2023-06-29:
imagine thinking I'd let anyone upload a random fucking zip file... :-1:

r4p7us, 2023-06-29:
no sorry, I just wanted to post an example nfc file to open an app with the pinball machine via nfc, that's all, since raw wouldn't let me send it, I zipped it.

ClaraCrazy, 2023-06-29:
use gist, please. gist.github.com

r4p7us, 2023-06-29:
okay sorry

Willy-JL, 2023-06-30:
@RogueMaster bounty time? kek

closed 2023-07-04 by Willy-JL