Flipper-XFW/Xtreme-Firmware #198: Enhancement: More consistant exit behaviour between apps

PowerUser64, 2023-04-10:
### Describe the enhancement you're suggesting.

There is a lot of variability in what you have to do to exit apps. For example, to exit the Bluetooth Remote, you have to press and hold the back button to bring up a menu that asks if you would like to exit or change remotes. Another example is some apps have a file manager that is opened at first when the app is opened, and when closing the file from the app, you are brought to the app selection screen (e.g. text viewer), whereas other apps bring you back to the file manager (e.g. music player). All this variability adds up, making some apps feel less cohesive with the experience of the rest of the system.

I would like to suggest that we update apps to follow a standard exit sequence.

As for what this exit sequence should be, in my experience, the apps that bring you back to the previous menu until you exit the app seem to be the easiest to use. If, for example, the Bluetooth Remote was updated to follow this sequence, it would mean removing the exit confirmation screen and having it bring you directly back to the device selection screen. To exit the app, you would just press back from the remote screen to get to the selection screen and then press back again to exit (the last step here is currently possible). This reduces the button presses required to exit the app and removes a screen from the app.

In other cases, like the text viewer, pressing back from a text file would bring you back to the file manager, selecting the file that was just used. This is consistent with the stock Flipper apps, like Sub GHz.

Additionally, for apps that use the file manager, pressing back from the start directory should return to the screen the file manager was opened from so you don't have to back out to the root directory to go back.

### Anything else?

I am new to the Flipper platform but not new to programming in C. Though I am not a master at C, I would like to work on implementing these changes in some of the apps I mentioned here, and possibly others if I get the hang of it.

Is this something that is feasible?

ClaraCrazy, 2023-04-10:
Doable, yes. but:

1. just not really my problem. Those apps are made by other people of the community
2. It would just cause trouble when pulling their new code, as that would revert my changes and then I'd have to do it again and again and again and.. you get the point

3. sending a PR to all of those... I'm totally into pain but thats too much, even for me lol

sowwy. still thanks for taking your time and suggesting this, and thank you for using our firmware <3

closed 2023-04-10 by ClaraCrazy

PowerUser64, 2023-04-10:
Makes sense. I'll ask some of the developers of these applications if they would be willing to make or accept changes like this. Thanks for your response!

PowerUser64, 2023-04-10:
@ClaraCrazy just fyi, there's a github action that keeps failing 😅
It sends me an email to tell me it failed each time I leave a comment :P

Here it is~

[image]