Flipper-XFW/Xtreme-Firmware #199: 125 khz - tag from electra intercom

salutcalin2002, 2023-04-10:
### Describe the enhancement you're suggesting.

Hello,
In Romania is a very common intercom company thatthe tag is specific to them, and the flipper zero can read them but cannot emulate at all-  the reader doesn't react to it .

there is a discussion on the forum Reddit
 https://forum.flipperzero.one/t/electra-intercom/6368
Not sure what we can do, or what information to bring, I can upload one of the tags

salutcalin2002, 2023-05-08:
Analyzing several dumps of the Electra tags, it seems they are adding 42 bits extra after the standard EM4100. So, you need a 128 bits card (e.g. an EM4200 that is backward compatible with older EM4100) in order to clone an Electra tag.

The raw emulation works as it’s not parsing the content in order to extract only the 64 bits (from which only 40bits are useful data, the rest being 9 ones header+10 rows and 4 columns parity bits+1 zero stop bit) but replays the original content.

salutcalin2002, 2023-05-08:
>: rfid raw_analyze /ext/lfrfid/RfidRecord.ask.raw
[32 131]        [32 99]
[316 511]       [316 195]
[316 512]       [316 196]
[316 510]       [316 194]
[317 511]       [317 194]
[317 511]       [317 194]
[317 511]       [317 194]
[316 511]       [316 195]
[317 512]       [317 195]
[316 510]       [316 194]
[317 511]       [317 194]
[317 512]       [317 195]
[315 511]       [315 196]
[316 510]       [316 194]
[316 511]       [316 195]
[316 511]       [316 195]
[315 511]       [315 196]
[316 511]       [316 195]
[315 510]       [315 195]
[316 510]       [316 194]
[316 510]       [316 194]
[315 511]       [315 196]
[572 767]       [572 195]
[316 511]       [316 195]
[316 511]       [316 195]
[316 511]       [316 195]
[316 511]       [316 195]
[315 511]       [315 196]
[315 511]       [315 196]
[315 511]       [315 196]
[315 791]       [315 476]
[288 487]       [288 199]
[314 511]       [314 197]
[314 511]       [314 197]
[314 511]       [314 197]
[314 511]       [314 197]
[315 511]       [315 196]
[570 767]       [570 197]
[314 791]       [314 477]
[547 743]       [547 196]
[315 511]       [315 196]
[315 791]       [315 476]
[547 743]       [547 196]
[316 790]       [316 474]
[289 487]       [289 198]
[315 511]       [315 196]
[571 1047]      [571 476]
[547 1022]      [547 475]
[549 1023]      [549 474]
[291 488]       [291 197]
[571 766]       [571 195]
[317 790]       [317 473]
[290 488]       [290 198]
[572 1046]      [572 474]
[291 488]       [291 197]
[316 510]       [316 194]
[573 1046]      [573 473]
[291 488]       [291 197]
[572 1046]      [572 474]
[550 743]       [550 193]
[318 511]       [318 193]
[318 791]       [318 473]
[291 487]       [291 196]
[316 510]       [316 194]
[574 1047]      [574 473]
[291 487]       [291 196]
[575 1047]      [575 472]
[550 1023]      [550 473]
[292 487]       [292 195]
[575 1046]      [575 471]        <FOUND EM4100>
   Frequency: 125000.000000
  Duty Cycle: 0.500000
       Warns: 0
   Pulse sum: 25554
Duration sum: 43597
     Average: 0.586141
    Protocol: EM4100 [03 E8 56 82 C9]
FC: 086, Card: 33481

salutcalin2002, 2023-05-08:
From a4736d395efaeb328849a2cf0b6327faa764bc34 Mon Sep 17 00:00:00 2001
From: Dan Caprita <dan@caprita.ro>
Date: Thu, 3 May 2023 22:29:09 +0300
Subject: [PATCH] [RFID] Add ELECTRA extra data: 0x7E1E[AAAAAAAAAAAA]

---
 lib/lfrfid/protocols/protocol_em4100.c | 10 ++++++++++
 1 file changed, 10 insertions(+)

diff --git a/lib/lfrfid/protocols/protocol_em4100.c b/lib/lfrfid/protocols/protocol_em4100.c
index 4b720dff..a3be54f2 100644
--- a/lib/lfrfid/protocols/protocol_em4100.c
+++ b/lib/lfrfid/protocols/protocol_em4100.c
@@ -35,10 +35,13 @@ typedef uint64_t EM4100DecodedData;
 #define EM_READ_LONG_TIME_LOW (EM_READ_LONG_TIME - EM_READ_JITTER_TIME)
 #define EM_READ_LONG_TIME_HIGH (EM_READ_LONG_TIME + EM_READ_JITTER_TIME)
.
+#define EM_ELECTRA_DATA 0x7E1EAAAAAAAAAAAA
+
 typedef struct {
     uint8_t data[EM4100_DECODED_DATA_SIZE];
.
     EM4100DecodedData encoded_data;
+    EM4100DecodedData encoded_data_old;
     uint8_t encoded_data_index;
     bool encoded_polarity;
.
@@ -237,6 +240,13 @@ LevelDuration protocol_em4100_encoder_yield(ProtocolEM4100* proto) {
         proto->encoded_polarity = true;
         proto->encoded_data_index++;
         if(proto->encoded_data_index >= 64) {
+            // alternate between actual data and Electra data
+            if (proto->encoded_data != EM_ELECTRA_DATA) {
+                proto->encoded_data_old = proto->encoded_data;
+                proto->encoded_data = EM_ELECTRA_DATA;
+            } else {
+                proto->encoded_data = proto->encoded_data_old;
+            }
             proto->encoded_data_index = 0;
         }
     }
--.
2.17.1

hp69ftz, 2023-05-25:
@ClaraCrazy https://forum.flipperzero.one/t/electra-intercom/6368/65

ClaraCrazy, 2023-05-25:
@Willy-JL

DerRumberg, 2023-08-22:
> Analyzing several dumps of the Electra tags, it seems they are adding 42 bits extra after the standard EM4100. So, you need a 128 bits card (e.g. an EM4200 that is backward compatible with older [...]

petruut, 2023-12-11:
Any news when this might make it into the firmware? I think a big part of Europe is full of these intercoms

novaventure, 2024-01-30:
@ClaraCrazy is the Electra feature going to be added soon to XFW? 🤞

VladFlorinIlie, 2024-05-18:
It seems that support for Electra has been added to the base Flipper firmware: https://github.com/flipperdevices/flipperzero-firmware/pull/3640

Could this be ported to Xtreme as well?

bogzik, 2024-05-19:
and in the fuzzer😅 tnks

VladFlorinIlie, 2024-05-19:
No need. I have switched to Momentum firmware. Xtreme seems to be pretty much dead anyway.

claudiuccg, 2024-06-11:
So it's working or not? it's added to the firmware ?