Flipper-XFW/Xtreme-Firmware #202: Mental Health 'App' Idea

BoomSec, 2023-04-11:
### Description of the feature you're suggesting.

**Mindfulness app?**
It may help for some mindfulness exercises to be available in the firmware
This make the Xtreme firmware really stand out. 

Making a plugin / app that you can open that walks you throw slow breathing exercises (with included flipper photos for extra happy chemicals). 

pseudo code start:

`display "breathe in pic / gif"`
`prompt user to breathe in,`
`wait 3-5 seconds`
`prompt user to hold breath for 1 second`
`wait 1 second`
`display "breathe out animation"`

pseudo code end

Possible options to click at your current breathing / heart rate, which the app would adapt to to give a more suitable breathing method to help ground the end user.

Maybe offer some other self care exercises / activities at another prompt??

If have any questions or want further elaboration / more ideas let me know

### Anything else?

Nothing further

ClaraCrazy, 2023-04-14:
quick reminder - this is a pentesting device

closed 2023-04-14 by ClaraCrazy

CapuchaRojo, 2023-04-14:
Howdy Clara

Remember me? I was dude from the discord. And I admittedly, was on a bender. Resulting in….  I annoyed u. Enough to punt me out the server.

Granted yes ok I get that. U in charge. So it is what it is.

But
I never stopped using extreme
I won’t. Loyal to a fault. Plus I’m no dummy. There is no doubt in my mind which firmware is the real deal Holyfield.
I need help. Bless everyone’s heart there. But I’m running your firmware. And I use iOS and arduino usually.
But marauder flashing onto my wifi dev unit in any of the ways available.. only lead to dead ends.  Closest I can get is it there but during the scans every bit of code is file bad header, repeated for 80 plus pages.
And i literally have had skeleton man try to walk me thru. And nah. I’m soup sammiching it still somehow.

U don’t have to help me

You might not even read this.

Fair

But if you do. Please know. I have learned my lesson C. Ain’t no way will I run amok in the chat and just chat as freely and as foolishly as I want. I realize I am not important. And in reality I need u all and absolutely not the other way around.

But C
I’m asking you simply
Please forgive my misdeeds and pls
Maybe on a trial basis even
Allow me back into the greatest collection of cyber revolutionaries that I’ve ever seen.

I admit I was wrong. And in no way does this justify. But Clara I had no idea. What it was like in the … elsewheres. Up to now even I’m not even at my 1 year anniversary on the discord. Only used it for Pokémon go raids before I got flipper and then met u and in a drug enduced spree I burnt some bridges rather quickly and I can say that I see the error of my ways. And being out here. Outside the garden of flippers’ Eden. Can get EXTREMELY frustrating.

I’ll pay some sort of punishment or something beneficial for the community

Or can I pay u cash?

I’m in Oregon do u need or want any alternative medicines? They are legal here. And such and such and

Do u need hardware?
Or can I start a charity in ur honor?
Do you like vacations? I have access to stupid deals.

Or may I create a Cynthia social media fan account, and have its “op host” (actually just chat bots) sell raffle tickets to such and such.
I can set up a cythnia only fans

Or pay for an affiliate thru jasper or wherever and get u passively paid.

These are just ideas. I’m open. And willing to accept anything hell or high water or whatever. I just wanna pay u back and above all else apologize to u directly.

Well
Maybe just maybe
U might just let me back into heaven

Please Clara
Please forgive me and allow me to come b home! I promise I will only use the strictest code of ethics in the open air chats. I won’t be the loud dumb ruggedly handsome yet slightly aloof American any longer ever again in the general and etc. I swear so help gods that my MacBook catch a virus.

Clara, this? It was good for me to pass the bad energy out. At the very least thank you for allowing me to atone there.

Thank u for ur considerations on all the rest,
🥂 Always

Adam V. M.
503-369-9283

Owner: A Storms Coming
Vaction bookings: Evolution Travel
Sent with Spark

somecoulombs, 2023-04-21:
https://cdn.discordapp.com/attachments/1056360523123408948/1099053010417619105/Adam_Mlady.mp3