Flipper-XFW/Xtreme-Firmware #205: Bluetooth Connection to phone device

Fernando144ft, 2023-04-13:
### Describe the bug.

I'm trying to connect the flipper zero with my phone, I did before I installed the Xtreme firmware, it must be a little bug maybe related with the name of the flipper. I have installed the latest release; done reboot to the flipper and clear the data on the phone app but it just keeps searching until it says "Flipper Devices Not Found".

### Reproduction

1. Turn Bluetooth Off
2. Turn Bluetooth On
3. Press Connect on the phone

Fernando144ft, 2023-04-13:
I just installed the default firmware to try connecting it with the phone and it worked, probably the firmware does have a bug with the Bluetooth.

Fernando144ft, 2023-04-13:
I reinstalled the Xtreme firmware and its bugging when it is trying to sync the flipper.

Lellosx90, 2023-04-17:
ciao, hai risolto? io non riesco ad usare Bad KB col bluetooth mi stacca la connessione .

cypheroxide, 2023-04-17:
From main screen animation (unlocked): Click the up arrow, go to settings (wrench and screwdriver icon), go to Bluetooth. Make sure this setting is set to on. If necessary, pair your flipper to your device again or use flipper on your computer.

FBReality, 2023-04-18:
> From main screen animation (unlocked): Click the up arrow, go to settings (wrench and screwdriver icon), go to Bluetooth. Make sure this setting is set to on. If necessary, pair your flipper to your device again or use qFlipper on your computer.

Same issue, updated firmware to latest, bluetooth is on into setting. Paired with my phone but app still try to connect

Ph4n70m-0v3rr1d3, 2023-04-21:
A work around that ive been using since v41 or so, 
1. unpair your flipper from your phone's bluetooth menu AND forget the device from the flipper mobile app.
2. forget all devices on your flipper in the bluetooth settings.
3. flash OFW
4. pair your flipper wi th your phone via the FLIPPER MOBILE APP
5. flash XFW

as long as you don't unpair your device from the flipper app or your phone you should be fine. your mileage may vary but this method has been working for me. afaik it's an ongoing investigation as to why this is happening. In the mean time give this work around a try and see if it works for you. 

Cheers!

Neodymium5312, 2023-05-01:
> A work around that ive been using since v41 or so,
>
> 1. unpair your flipper from your phone's bluetooth menu AND forget the device from the flipper mobile app.
> 2. forget all devices on your flipper [...]

Tis also works if you flash rogue master before pairing and then flash back to Xtreme firmware

bosund, 2023-05-04:
I have the same issue.

ClaraCrazy, 2023-05-04:
I got 3 devices over here, WillyJL has 2, and nobody can replicate it. 

Unless someone finally sends some logs one could use and juat says "issue" like some mindless robot sent by god-knows who, this will be closed bc its impossible to replicate on Android **or** IOS from even a single person of the team or close community. 

Thanks.

bosund, 2023-05-04:
I can replicate it every time. I can't see Flipper from the app, so it is impossible to upload logfiles.

I can connect from my ipad but not from my Android phone.
I think it is related to android 13 april update. I have a Pixel 6 pro.

ScootinCivic, 2023-05-04:
Same here as well.
Running Android 11

HOWEVER ~ I am able to connect via phone (stand alone).
However the app doesn't see the zero at all. No matter how I may try to connect.

NOTE: I ran the OS deva build then flashed the current release.

🤷

jl1990, 2023-05-05:
I was able to replicate it in my device.

Looks like the application can find the device if using the default name.
When you change the flipper name with Xtreme settings, it won't find the device.

This was the cause for me

ClaraCrazy, 2023-05-05:
@FBReality @Neodymium5312 @bosund @Fernando144ft please run `info device` in Apps -> Tools -> CLI

Report back the following values: `hardware.uid` and hardware.timestamp`

This will not just help fix this, but also relates to an ongoing investigation with Flipper Zero that seems to affect many devices. Thank you for your cooperation

ClaraCrazy, 2023-05-05:
forgot some pings. @Lellosx90 @Ph4n70m-0v3rr1d3 @cypheroxide @ScootinCivic

Neodymium5312, 2023-05-05:
Hardware timestamp: 1678420824
Hardware device: B3629D0027E18000
Also as an update latest Xtreme firmware is having same issue. I flashed official release version and connected to phone app successfully. And after flashing Xtreme firmware device is able to connect and sync to phone but I can no longer trigger Bluetooth remotes for subghz. I also noticed the radio stack version is different from official whether this is related or now I'm not sure. @ClaraCrazy if you need anything else I will be willing to help however I can

ClaraCrazy, 2023-05-05:
also, all of you pinged ones... please update to latest, turn off debug, let it idle for a few hours and let me know if your flipper crashes eventually. 

We believe to have found an issue with the device itself, causing random crashes and your connection issues.

Neodymium5312, 2023-05-05:
> also, all of you pinged ones... please update to latest, turn off debug, let it idle for a few hours and let me know if your flipper crashes eventually.
> 
> We believe to have found an issue with the device itself, causing random crashes and your connection issues.

Just to clarify, update to latest Xtreme firmware?

ClaraCrazy, 2023-05-05:
yes. fixed wording

bosund, 2023-05-05:
Hardware.timestamp 677670740
Hardware.uid A00027E18000

Running latest Xtreme firmware
No crashes here

ScootinCivic, 2023-05-05:
Hardware.timestamp 1680790548
Hardware.uid B57CB60027E18000

No crashes here either

Neodymium5312, 2023-05-06:
I've been idling for hours now with no crashing. Is there anything else we would need to set to a certain value besides the debug option? I've modified quite a few of the system settings so maybe something else is toggles that is preventing me from replicating the crash

ScootinCivic, 2023-05-06:
After everything I tryed the following & I was able to connect..

Change your flippers name to just Flipper ('Flipper').

-Xtreme-Misc-Change Device Name

Just noticed my other OG flipper had the name "Flipper Xxxxxx"..

Anyways try that.

ClaraCrazy, 2023-05-06:
its a bad solution. just wait for the update

Neodymium5312, 2023-05-06:
> After everything I tryed the following & I was able to connect..
> 
> Change your flippers name to just Flipper ('Flipper').
> 
> -Xtreme-Misc-Change Device Name
> 
> Just noticed my other OG flipper had the name "Flipper Xxxxxx"..
> 
> Anyways try that.

I've already tested renaming the device different things as well as using the default device name. Neither fixes the initial pairing or the ability to use remote triggers for subghz

closed 2023-05-06 by github-actions[bot]