Flipper-XFW/Xtreme-Firmware #207: SubGHz Read RAW glitch

MaxOutStudio, 2023-04-14:
### Describe the bug.

When using Firmware version "XFW-0044_09042023" the Read RAW works once when receiving and sending signals, but if you erase it while viewing it, then try and scan another signal and send it again, it shows it being sent but it doesn't replay the signal to the targeted device at all, and normally when recording a signal and sending it, it lasts the same amount of time you record it for then ends automatically, the second time as mentioned above happens but it also stays in the sending signal animation loop until you end it yourself. Not sure what's going on, but this is my report on the details about the bug.

### Reproduction

1. Enter SubGHZ in the main menu of the Flipper
2. Click "Read RAW"
3. Record a signal, any signal that you can replay on a target device that will show signs of receiving it.
4. Send the signal
5. Once the target device has shown signs that it has received the recorded signal, erase the signal and record the same one or a new one
6. try sending the same signal and you should hopefully you should see that the targeted device hasn't received the recorded RAW signal and stays in the sending animation loop.

### Logs

```Text
37263 [D][FuriHalSubGhz] Loading custom preset
37265 [D][FuriHalSubGhz] Reg[0]: 02=0D
37267 [D][FuriHalSubGhz] Reg[2]: 03=07
37269 [D][FuriHalSubGhz] Reg[4]: 08=32
37271 [D][FuriHalSubGhz] Reg[6]: 0B=06
37272 [D][FuriHalSubGhz] Reg[8]: 14=00
37275 [D][FuriHalSubGhz] Reg[10]: 13=00
37277 [D][FuriHalSubGhz] Reg[12]: 12=30
37279 [D][FuriHalSubGhz] Reg[14]: 11=32
37281 [D][FuriHalSubGhz] Reg[16]: 10=17
37283 [D][FuriHalSubGhz] Reg[18]: 18=18
37285 [D][FuriHalSubGhz] Reg[20]: 19=18
37287 [D][FuriHalSubGhz] Reg[22]: 1D=91
37289 [D][FuriHalSubGhz] Reg[24]: 1C=00
37291 [D][FuriHalSubGhz] Reg[26]: 1B=07
37293 [D][FuriHalSubGhz] Reg[28]: 20=FB
37295 [D][FuriHalSubGhz] Reg[30]: 22=11
37297 [D][FuriHalSubGhz] Reg[32]: 21=B6
37299 [D][FuriHalSubGhz] PA[34]:  00
37300 [D][FuriHalSubGhz] PA[35]:  00
37303 [D][FuriHalSubGhz] PA[36]:  00
37304 [D][FuriHalSubGhz] PA[37]:  C0
37307 [D][FuriHalSubGhz] PA[38]:  00
37309 [D][FuriHalSubGhz] PA[39]:  00
37311 [D][FuriHalSubGhz] PA[40]:  00
37313 [D][FuriHalSubGhz] PA[41]:  00
37315 [D][FuriHalSubGhz] PA[42]:  00
37317 [D][FuriHalSubGhz] PA[43]:  00
37320 [D][DolphinState] icounter 674, butthurt 0
42705 [I][SubGhzFileEncoderWorker] Worker start
42726 [I][SubGhzFileEncoderWorker] Start transmission
42796 [I][SubGhzFileEncoderWorker] End read file
42805 [D][FuriHalSubGhz] Loading custom preset
42809 [D][FuriHalSubGhz] Reg[0]: 02=0D
42811 [D][FuriHalSubGhz] Reg[2]: 03=07
42813 [D][FuriHalSubGhz] Reg[4]: 08=32
42815 [D][FuriHalSubGhz] Reg[6]: 0B=06
42817 [D][FuriHalSubGhz] Reg[8]: 14=00
42819 [D][FuriHalSubGhz] Reg[10]: 13=00
42821 [D][FuriHalSubGhz] Reg[12]: 12=30
42823 [D][FuriHalSubGhz] Reg[14]: 11=32
42825 [D][FuriHalSubGhz] Reg[16]: 10=17
42829 [D][FuriHalSubGhz] Reg[18]: 18=18
42831 [D][FuriHalSubGhz] Reg[20]: 19=18
42833 [D][FuriHalSubGhz] Reg[22]: 1D=91
42835 [D][FuriHalSubGhz] Reg[24]: 1C=00
42837 [D][FuriHalSubGhz] Reg[26]: 1B=07
42839 [D][FuriHalSubGhz] Reg[28]: 20=FB
42841 [D][FuriHalSubGhz] Reg[30]: 22=11
42843 [D][FuriHalSubGhz] Reg[32]: 21=B6
42845 [D][FuriHalSubGhz] PA[34]:  00
42846 [D][FuriHalSubGhz] PA[35]:  00
42849 [D][FuriHalSubGhz] PA[36]:  00
42851 [D][FuriHalSubGhz] PA[37]:  C0
42853 [D][FuriHalSubGhz] PA[38]:  00
42855 [D][FuriHalSubGhz] PA[39]:  00
42857 [D][FuriHalSubGhz] PA[40]:  00
42859 [D][FuriHalSubGhz] PA[41]:  00
42861 [D][FuriHalSubGhz] PA[42]:  00
42863 [D][FuriHalSubGhz] PA[43]:  00
46230 [I][SubGhzFileEncoderWorker] End transmission
46232 [D][FuriHalSubGhz] Async TX Radio stats: on 2277726us, off 1077826us, DutyCycle: 68%
46283 [I][SubGhzFileEncoderWorker] Worker stop
52328 [D][DolphinState] icounter 674, butthurt 0
52331 [D][FuriHalSubGhz] Loading custom preset
52333 [D][FuriHalSubGhz] Reg[0]: 02=0D
52335 [D][FuriHalSubGhz] Reg[2]: 03=07
52337 [D][FuriHalSubGhz] Reg[4]: 08=32
52339 [D][FuriHalSubGhz] Reg[6]: 0B=06
52341 [D][FuriHalSubGhz] Reg[8]: 14=00
52343 [D][FuriHalSubGhz] Reg[10]: 13=00
52344 [D][FuriHalSubGhz] Reg[12]: 12=30
52347 [D][FuriHalSubGhz] Reg[14]: 11=32
52349 [D][FuriHalSubGhz] Reg[16]: 10=17
52350 [D][FuriHalSubGhz] Reg[18]: 18=18
52353 [D][FuriHalSubGhz] Reg[20]: 19=18
52355 [D][FuriHalSubGhz] Reg[22]: 1D=91
52357 [D][FuriHalSubGhz] Reg[24]: 1C=00
52359 [D][FuriHalSubGhz] Reg[26]: 1B=07
52361 [D][FuriHalSubGhz] Reg[28]: 20=FB
52363 [D][FuriHalSubGhz] Reg[30]: 22=11
52365 [D][FuriHalSubGhz] Reg[32]: 21=B6
52367 [D][FuriHalSubGhz] PA[34]:  00
52369 [D][FuriHalSubGhz] PA[35]:  00
52371 [D][FuriHalSubGhz] PA[36]:  00
52373 [D][FuriHalSubGhz] PA[37]:  C0
52375 [D][FuriHalSubGhz] PA[38]:  00
52377 [D][FuriHalSubGhz] PA[39]:  00
52379 [D][FuriHalSubGhz] PA[40]:  00
52381 [D][FuriHalSubGhz] PA[41]:  00
52383 [D][FuriHalSubGhz] PA[42]:  00
52385 [D][FuriHalSubGhz] PA[43]:  00
57092 [I][SubGhzFileEncoderWorker] Worker start
57113 [I][SubGhzFileEncoderWorker] Start transmission
57192 [D][FuriHalSubGhz] Loading custom preset
57194 [D][FuriHalSubGhz] Reg[0]: 02=0D
57196 [D][FuriHalSubGhz] Reg[2]: 03=07
57197 [D][FuriHalSubGhz] Reg[4]: 08=32
57199 [D][FuriHalSubGhz] Reg[6]: 0B=06
57201 [D][FuriHalSubGhz] Reg[8]: 14=00
57203 [D][FuriHalSubGhz] Reg[10]: 13=00
57205 [D][FuriHalSubGhz] Reg[12]: 12=30
57207 [I][SubGhzFileEncoderWorker] End read file
57209 [D][FuriHalSubGhz] Reg[14]: 11=32
57211 [D][FuriHalSubGhz] Reg[16]: 10=17
57213 [D][FuriHalSubGhz] Reg[18]: 18=18
57215 [D][FuriHalSubGhz] Reg[20]: 19=18
57217 [D][FuriHalSubGhz] Reg[22]: 1D=91
57219 [D][FuriHalSubGhz] Reg[24]: 1C=00
57221 [D][FuriHalSubGhz] Reg[26]: 1B=07
57223 [D][FuriHalSubGhz] Reg[28]: 20=FB
57225 [D][FuriHalSubGhz] Reg[30]: 22=11
57227 [D][FuriHalSubGhz] Reg[32]: 21=B6
57229 [D][FuriHalSubGhz] PA[34]:  00
57231 [D][FuriHalSubGhz] PA[35]:  00
57232 [D][FuriHalSubGhz] PA[36]:  00
57234 [D][FuriHalSubGhz] PA[37]:  C0
57235 [D][FuriHalSubGhz] PA[38]:  00
57237 [D][FuriHalSubGhz] PA[39]:  00
57238 [D][FuriHalSubGhz] PA[40]:  00
57240 [D][FuriHalSubGhz] PA[41]:  00
57242 [D][FuriHalSubGhz] PA[42]:  00
57244 [D][FuriHalSubGhz] PA[43]:  00
70616 [D][FuriHalSubGhz] Async TX Radio stats: on 675071us, off 1e+09us, DutyCycle: 0%
70622 [I][SubGhzFileEncoderWorker] End transmission
70626 [I][SubGhzFileEncoderWorker] Worker stop
```

Willy-JL, 2023-04-28:
Can't replicate, neither on public v44 release nor on current dev branch.

Are you positive this is reproducible and not just a fluke? Any other info you can give?

MaxOutStudio, 2023-04-30:
It was indeed a fluke, I apologize for the confusion and I will close this issue now.

closed 2023-04-30 by MaxOutStudio