Flipper-XFW/Xtreme-Firmware #209: Can't reset PIN

iMoses, 2023-04-18:
### Describe the bug.

Using the official method for reseting the pin, [as described here](https://docs.flipperzero.one/basics/firmware-update/firmware-recovery), only works on the original firmware. It seems that each version has its own reference for a PIN password.

### Reproduction

1. Set PIN
2. Follow [guide](https://docs.flipperzero.one/basics/firmware-update/firmware-recovery)
3. Change PIN (works on original FW)
4. Install Flipper-Xtreme
5. PIN is still the old one, new PIN doesn't work

ClaraCrazy, 2023-04-19:
Not a bug, and not even how to reset a pin....

https://docs.flipperzero.one/basics/control#c_9ya

closed 2023-04-19 by ClaraCrazy

ggppjj, 2023-04-27:
Hello!

I was attempting to follow the _official_ instructions for a pin/factory reset, but my flipper kept crashing and never showing any confirmation page. Is this something that would be broken on just my device somehow?

I had to flash the dirty dfu from the link above and delete the .config directory as the OP did in order to reset.