Flipper-XFW/Xtreme-Firmware #210: Change RFID and NFC app name to LF RFID and HF RFID. The current one is misleading. Both are RFID.

G0ldi2222, 2023-04-20:
### Describe the enhancement you're suggesting.

Description speaks for itself. Both apps are RFID apps and NFC is just a protocol (which doesn’t included locate cards but can be used on migrate cards as well. So I suggest to change them to Low Frequency RFID (125khz and 136khz) and Hight Frequency RFID (13,56mhz) - ergo LF RFID and HF RFID. Itbwiöö help a lot to understand the difference instead of mislead.

Ph4n70m-0v3rr1d3, 2023-04-20:
As an amateur radio license holder in the USA, I feel like it's arbitrary and unnecessary. If we're gonna split hairs, I'd recommend calling NFC "HF 10cm band RFID @ 13.56mhz" and call the other "LF 2400m band RFID @ 25khz". This way we can without a shadow of a doubt make the descriptions retard proof.... 

but seriously, if you dont know what NFC is to begin with, then you should either A do some research about the device and its functions, or B sell your flipper cause you probabaly shouldn't have one... i wasn't misled at all when i scanned NFC with my flipper. i knew it was scanning 13.56 mhz. jus sayin.....

ClaraCrazy, 2023-04-20:
Some people, I swear...

closed 2023-04-20 by ClaraCrazy

G0ldi2222, 2023-04-22:
> As an amateur radio license holder in the USA, I feel like it's arbitrary and unnecessary. If we're gonna split hairs, I'd recommend calling NFC "HF 10cm band RFID @ 13.56mhz" and call the other "LF [...]

clearly an amatuer and an typical american who dont want others to be educated and rather distribure false information (25khz??? It is 125-134KHZ - AMATEUR)! So to explain it to someone who clearly needs a retard proof explanation about those two, "Every NFC device is infact an RFID device, but not every RFID device is automatically and NFC device!"

Ph4n70m-0v3rr1d3, 2023-04-22:
Oh man, God forbid somebody types a typo on their cellphone. I'm sure that's never happened before in the history of cellphone usage lmao. You're just salty cause your idea is dumb. Your idea is bad and you should feel bad.

Ph4n70m-0v3rr1d3, 2023-04-22:
All NFC devices use RFID Signals. But not all RFID Signals communicate with NFC chips. Easy enough to understand yes?

G0ldi2222, 2023-04-22:
> All NFC devices use RFID Signals. But not all RFID Signals communicate with NFC chips. Easy enough to understand yes?

exactly what i said.. finally you understood it as well... so yeah.. lets all use an educational device to mislead and uneducate the others, who dont know the difference... american stupity at its best

Ph4n70m-0v3rr1d3, 2023-04-22:
I'd rather be an American than be from wherever you live. Clearly their education system is non existent. Lol. Either that or your brain is a bit too smooth.... Your family tree has a few straight branches if ya catch my drift... You're a few fries short of a happy meal.. etc.

Ph4n70m-0v3rr1d3, 2023-04-22:
I grow tired of this little back and forth. I don't argue with people who just throw random insults based on where they're from or what gender they are or how tall or fat they are. It shows how low their IQ is when they can't even hold a conversation without throwing insults when losing an argument. So welcome to my block list. Enjoy your stay.