Flipper-XFW/Xtreme-Firmware #212: I need some help to transfer the badkb function to my custom fw

UnsavorySpirit, 2023-04-24:
### Describe the bug.

i need some help because when i tried building the fw i got this error
### PS H:\flipper zero\unleashed-firmware> ./fbt
        ICONS   build\f7-firmware-D\assets\compiled\assets_icons.c
        PROTO   assets\protobuf\application.proto
        DOLPHIN blocking
        DOLPHIN internal
2023-04-24 14:16:50,992 [INFO] Processing Dolphin sources
2023-04-24 14:16:50,992 [INFO] Loading data
2023-04-24 14:16:50,992 [INFO] Loading directory assets\dolphin\blocking
2023-04-24 14:16:50,993 [INFO] Loading meta from assets\dolphin\blocking\L0_NoDb_128x51\meta.txt
        APPS    build\f7-firmware-D\applications\applications.c
2023-04-24 14:16:50,994 [INFO] Loading meta from assets\dolphin\blocking\L0_SdBad_128x51\meta.txt
2023-04-24 14:16:50,994 [INFO] Loading meta from assets\dolphin\blocking\L0_SdOk_128x51\meta.txt
2023-04-24 14:16:50,995 [INFO] Loading meta from assets\dolphin\blocking\L0_Url_128x51\meta.txt
2023-04-24 14:16:50,996 [INFO] Loading meta from assets\dolphin\blocking\L0_NewMail_128x51\meta.txt
2023-04-24 14:16:50,996 [INFO] Packing
2023-04-24 14:16:51,007 [INFO] Processing Dolphin sources
2023-04-24 14:16:51,007 [INFO] Loading data
2023-04-24 14:16:51,007 [INFO] Loading directory assets\dolphin\internal
2023-04-24 14:16:51,008 [INFO] Loading meta from assets\dolphin\internal\L1_Tv_128x47\meta.txt
2023-04-24 14:16:51,009 [INFO] Loading meta from assets\dolphin\internal\L1_BadBattery_128x47\meta.txt
2023-04-24 14:16:51,009 [INFO] Loading meta from assets\dolphin\internal\L1_NoSd_128x49\meta.txt
2023-04-24 14:16:51,010 [INFO] Packing
        CC      applications\main\archive\helpers\archive_apps.c
        CC      applications\main\archive\helpers\archive_browser.c
        CC      applications\main\archive\helpers\archive_favorites.c
        CC      applications\main\archive\helpers\archive_files.c
2023-04-24 14:16:51,165 [INFO] Complete
2023-04-24 14:16:51,174 [INFO] Complete
        CC      applications\main\archive\scenes\archive_scene.c
        CC      applications\main\archive\scenes\archive_scene_browser.c
        CC      build\f7-firmware-D\applications\applications.c
        CC      applications\main\archive\scenes\archive_scene_delete.c
        PBVER   build\f7-firmware-D\assets\compiled\protobuf_version.h
        CC      applications\main\archive\scenes\archive_scene_info.c
        CC      applications\main\archive\scenes\archive_scene_rename.c
        CC      applications\main\archive\scenes\archive_scene_show.c
        CC      applications\main\archive\views\archive_browser_view.c
        APIPREP build\f7-firmware-D\sdk_origin.i
        APIPP   build\f7-firmware-D\sdk_origin.i
        CC      applications\main\archive\archive.c
In file included from applications\main\archive\views\archive_browser_view.c:1:
build\f7-firmware-D\assets\compiled/assets_icons.h:46:19: error: redundant redeclaration of 'I_Clock_18x18' [-Werror=redundant-decls]
   46 | extern const Icon I_Clock_18x18;
      |                   ^~~~~~~~~~~~~
build\f7-firmware-D\assets\compiled/assets_icons.h:37:19: note: previous declaration of 'I_Clock_18x18' was here
   37 | extern const Icon I_Clock_18x18;
      |                   ^~~~~~~~~~~~~
build\f7-firmware-D\assets\compiled/assets_icons.h:47:19: error: redundant redeclaration of 'I_Error_18x18' [-Werror=redundant-decls]
   47 | extern const Icon I_Error_18x18;
      |                   ^~~~~~~~~~~~~
build\f7-firmware-D\assets\compiled/assets_icons.h:38:19: note: previous declaration of 'I_Error_18x18' was here
   38 | extern const Icon I_Error_18x18;
      |                   ^~~~~~~~~~~~~
build\f7-firmware-D\assets\compiled/assets_icons.h:48:19: error: redundant redeclaration of 'I_EviSmile1_18x21' [-Werror=redundant-decls]
   48 | extern const Icon I_EviSmile1_18x21;
      |                   ^~~~~~~~~~~~~~~~~
build\f7-firmware-D\assets\compiled/assets_icons.h:39:19: note: previous declaration of 'I_EviSmile1_18x21' was here
   39 | extern const Icon I_EviSmile1_18x21;
      |                   ^~~~~~~~~~~~~~~~~
build\f7-firmware-D\assets\compiled/assets_icons.h:49:19: error: redundant redeclaration of 'I_EviSmile2_18x21' [-Werror=redundant-decls]
   49 | extern const Icon I_EviSmile2_18x21;
      |                   ^~~~~~~~~~~~~~~~~
build\f7-firmware-D\assets\compiled/assets_icons.h:40:19: note: previous declaration of 'I_EviSmile2_18x21' was here
   40 | extern const Icon I_EviSmile2_18x21;
      |                   ^~~~~~~~~~~~~~~~~
build\f7-firmware-D\assets\compiled/assets_icons.h:50:19: error: redundant redeclaration of 'I_EviWaiting1_18x21' [-Werror=redundant-decls]
   50 | extern const Icon I_EviWaiting1_18x21;
[... 47 lines cut ...]
applications\main\bad_kb\helpers\ducky_script.c:554:30: error: 'DolphinDeedBadKbPlayScript' undeclared (first use in this function); did you mean 'DolphinDeedBadUsbPlayScript'?
  554 |                 DOLPHIN_DEED(   CC      applications\main\bad_kb\scenes\bad_kb_scene.c
DolphinDeedBadKbPlayScript);
      |                              ^~~~~~~~~~~~~~~~~~~~~~~~~~
applications\services/dolphin/dolphin.h:32:31: note: in definition of macro 'DOLPHIN_DEED'
   32 |         dolphin_deed(dolphin, deed);                              \
      |                               ^~~~
applications\main\bad_kb\helpers\ducky_script.c:554:30: note: each undeclared identifier is reported only once for each function it appears in
  554 |                 DOLPHIN_DEED(DolphinDeedBadKbPlayScript);
      |                              ^~~~~~~~~~~~~~~~~~~~~~~~~~
applications\services/dolphin/dolphin.h:32:31: note: in definition of macro 'DOLPHIN_DEED'
   32 |         dolphin_deed(dolphin, deed);                              \
      |                               ^~~~
cc1.exe: all warnings being treated as errors
scons: *** [build\f7-firmware-D\applications\main\bad_kb\helpers\ducky_script.o] Error 1
cc1.exe: all warnings being treated as errors
scons: *** [build\f7-firmware-D\applications\main\archive\views\archive_browser_view.o] Error 1

********** FBT ERRORS **********
build\f7-firmware-D\applications\main\bad_kb\helpers\ducky_script.o: Error 1
build\f7-firmware-D\applications\main\archive\views\archive_browser_view.o: Error 1

### Reproduction

./fbt

ClaraCrazy, 2023-04-24:
It only works with xtreme, as it uses our custom api, xtreme functions etc. Nano will confirm this, as I am in close contact with him for development of this. 

With all respect, why are you even using a different firmware? We have every single existing feature of other firmware, and more (multiple new nfc protocols, tons of customization etc)

closed 2023-04-24 by UnsavorySpirit

ClaraCrazy, 2023-04-24:
That was not an answer to my question, but sure. You're... welcome, I guess?

UnsavorySpirit, 2023-04-24:
i'm just trying to build my own fw for learning
That's the reason

ClaraCrazy, 2023-04-24:
Well, thats always welcome. Should join our discord, we're happy to teach new aspiring devs. Not trying to gatekeep or anything. But perhaps you should start with something simpler, that doesnt require half of the firmware core to be ported haha

UnsavorySpirit, 2023-04-24:
ok thx

Tectectonic, 2023-04-24:
clara omg i miss the server so bad man im so disappointed in some of you guys i really wanna come back!