Flipper-XFW/Xtreme-Firmware #214: unreasonable ban

Tectectonic, 2023-04-24:
bru i was in the flipper unleashed discord and this dude starts tripping on me for no reason because i apparently was using general chat wrong and i told him why i did what i did (i literally said general chat is usually for general discussion) and then he started saying i was arguing when i wasn't then he used my username wrong so i corrected him and he banned me like wtf? what rules did i break? I `then went to the extreme server and was like "wtf i got banned for almost no reason then this guy proceeds to ban me again before i even did anything like all i said was i was disappointed in him for his weird behavior like bro what? i can count on 1 hand how mant discord servers i have been banned from (less than 3) and this is by far the most ridiculous reason. i got no ban message and i got no warning or nothing like bru what? please unban me flipper xtreme mods at least so i can share and receive sub files from the server and you don't even have to punish the mod i jus wanna get unbanned i wont even communicate in server just collect files.

Tectectonic, 2023-04-24:
but yeah idk he probably made a mistake or was having a bad day or sum but i aint break no rules so idk if i shoulda really been banned

ClaraCrazy, 2023-04-25:
done

closed 2023-04-25 by ClaraCrazy

Tectectonic, 2023-04-25:
thanks clara could you please send a invite back to server cus i lost it i think

CapuchaRojo, 2023-04-25:
Hi. Same issue. Can’t get into the server and need help with several of the pieces I have not being able to use after flash.. Can I get back in you? Please

Adam V. M.
503-369-9283

Owner: A Storms Coming
Vaction bookings: Evolution Travel
Sent with Spark